micro-manager/micro-manager #685: Datastore silently accepts images of different dimensions

nicost, 2020-01-31:
Using the putImage function on a datastore and sending images of different sizes does not cause any error whatsoever. This can lead to difficult to trace failures. I think that it would be best for the putImage function to throw an exception.

Illustrated with the following script:

targetStore = mm.data().createRAMDatastore();
mm.displays().createDisplay(targetStore);

nrT = 3;
cb = Coordinates.builder();
cb.c(0).t(0).p(0).z(0);

for (int t =0; t < nrT; t++) {
	cb.t(t);
	imgs = mm.live().snap(true);
	img = imgs.get(0).copyAtCoords(cb.build());
	targetStore.putImage(img);
	xSize = Integer.parseInt(mmc.getProperty("Camera", "OnCameraCCDXSize"));
	xSize--;
	mmc.setProperty("Camera", "OnCameraCCDXSize", xSize);
}

marktsuchida, 2020-02-12:
For the file-backed datastores, an exception should definitely be raised. 

For RAM datastores, I think the original intent was to allow planes of different size, although I kind of like the cleanness of just forbidding it. I wouldn't be opposed to throwing an exception and seeing if anybody complains. Alternatively, heterogeneous plane sizes could be made an optional (off-by-default) feature of the RAM datastore. One thing to check is whether the Snap/Live datastore depends on the current behavior when the camera image size changes.

nicost, 2020-04-20:
Throwing an exception is an api change and will work its way through most of the code and plugins.  I'll start work on a PR unless @marktsuchida things this will come up in the api related work

PR #1200 (2021-04-28): Adds ImagesDifferInSizeExceptions thrown by Storage implementations.
Closes #685.
closed 2021-04-28 by nicost