micro-manager/micro-manager #689: Request: Flush log file.

nanthony21, 2020-02-04:
There are many cases when testing that it would be nice to view the Core Log file while you're in the middle of doing something, however the log file doesn't seems to get written to disk until you close Micro-Manager. It would be nice if Micro-Manager would occasionally `flush` the log file to disk to force it to write.

I looked into doing this myself but quickly found myself deep in the C++ core of Micro-Manager. Maybe someone more familiar with the logging system would be willing to do this?

marktsuchida, 2020-02-12:
I wrote the logging code in MMCore. The file should be flushed quite often (10 ms when there are log entries to write): https://github.com/micro-manager/micro-manager/blob/2bf3702dae0fad889da5ee8c3d8f656ec6a843bc/MMCore/Logging/GenericPacketQueue.h#L130-L131 https://github.com/micro-manager/micro-manager/blob/2bf3702dae0fad889da5ee8c3d8f656ec6a843bc/MMCore/Logging/GenericStreamSink.h#L153

On Windows I routinely use a program called [SnakeTail](http://snakenest.com/snaketail/) to view the core log, and it displays in real time (as far as one can visually tell). On Mac/Linux the `tail -f` command should do the same.

There may be a delay if one of the other log destinations (such as standard error) is a slow device (such as Windows Command Prompt), because all destinations are written at the same pace. With Mac and Linux terminals this should not normally be an issue.

If you do not see the log updating in near real time, please provide us with more details to reproduce the issue.

Hint: you can generate single log messages manually from the Problem Report, for testing. Turn off "crash-robust logging" to test under the same conditions as when not running a Problem Report. (Crash-robust logging corresponds to the `synchronous` flag passed to `CMMCore::startSecondaryLogFile()`.)

nanthony21, 2020-02-12:
@marktsuchida Thanks for the help, It appears that this was actually more of an issue with notepad++ which I usually use.

While notepad++ usually detects when an opened file has changed, for some reason it doesn't work for the micro-manager log until micro-manager is closed. Even reopening the log file didn't work unless I shut down notepad ++ entirely and restarted it. Snaketail works perfectly though.

closed 2020-02-12 by nanthony21