micro-manager/micro-manager #690: Position list `Grid` metadata is lost upon loading

nanthony21, 2020-02-04:
If you generate a grid of positions using the `Tile Creator` dialog it will save some useful           "GridCol" and "GridRow" metadata with the positions. However, if you save these positions to file and then load the positions again you will find that this metadata has all been set to 0.

nanthony21, 2020-02-04:
I have tracked this problem down to the `DefaultPropertyMap.containsNumber` method. Call stack below:

DefaultPropertyMap.containsNumber:274	
DefaultPropertyMap.getAsNumber:280	
MultiStagePosition.fromPropertyMap:450	
PositionList.replaceWithPropertyMap:262	
PositionList.load:341	

The `GridRow` and `GridCol` items of the property map are loaded as `Integer` type but `containsNumber` only checks for primitives (`int`, `double`, etc.) This causes `getAsNumber` to fall back on the default value.

I'm not sure what the best way to proceed is: 1: Change `containsNumber` to also account for boxed datatypes or 2: change the `MultiStagePosition` class to save as `int` rather than `Integer`.

While the `DefaultPropertyMap` class is used for many different things it looks like `getAsNumber` and `containsNumber` are only used for the `GridRow` and `GridCol` of `MultiStagePosition`, this explains why this bug has gone unnoticed. Maybe it would be best to deprecate these methods and just use the more commonly used `getInteger` method?

nicost, 2020-02-04:
Thanks for your thorough analysis.  I think that reading the values back in as Integers is the most backward compatible fix (and also fine in general), so I applied that fix.

closed 2020-02-04 by nicost