micro-manager/micro-manager #694: live view randomly stops

andybra, 2020-02-10:
we have switched to MM2 gamma and with one the later nightly builds the live view often just freezes/stops. Only stopping and restarting live helps. we consider this quite critical, since when looking through the oculars and trying to add some images to albums one now needs to double check each acquisition to make sure the software can add an image to album, therefore. This then can easily cause a lot of extra work, having to redo some acquisitions. 
I'm aware there was such an issue years ago, but I've never discovered it until recently.
The camera is from PCO, it doesn't stall in other applications.

nicost, 2020-02-10:
Did it work correctly in earlier nightly builds of MM2-gamma?  If so, it would help immensely if you could pinpoint at which specific date this problem occurred.   
Did you set the Sequence buffer size (Tools > Options) to the same size that you used in 1.4?
It would be helpful to generate a Trouble Report (Help  > Report a Porblem) that catures the live mode stopping, and send it.

closed 2020-02-10 by nicost

andybra, 2020-02-11:
sorry, I can't provide the info which build broke it. We hadn't followed the updates this accurately. The error occurs 'maybe' after 5-10 minutes of usage and we can't reproduce it, so it'd take some time to find out. I'll see if I can find the time to at least home in on the exact date.
I'm quite confident that it didn't occur with laser years builds (end of year). Since then we updated maybe 2 times.
All our systems are (and need to be) offline, I will see if I can provide the Problem-Report by copy/paste. All our lab equipment is offline, that's why we usually never even consider such functionality. Also keeping stuff running without being able to update it other than carrying around USB-Sticks is quite ineffective. :-/  

I'll follow up on this