micro-manager/micro-manager #697: Setting of pixel size is unintuitive

nanthony21, 2020-02-11:
This actually a collection of issues that make pixel size calibration a bit confusing.

- The pixel size calibration window includes a field for the pixel size as well as fields for a 2x2 affine transformation. An unsuspecting user who doesn't use the automated pixel calibration routine may enter just the pixel size and assume that they can leave the affine transformation as [[1, 0], [0, 1]] since there is no explanation of how the affine transform is scaled. However in order for the calibration to actually be effective they need to hit the "Calculate" so the the matrix looks like [[pixSize, 0], [0, pixSize]]

- Closing the pixel size calibration window after making changes should ask the user if they want to save the configuration, many users will not understand that unlike many other settings in Micro-Manager this information is not saved automatically.

- Even once a user has saved the correct affine transform to the configuration file, MMCore.getPixelSizeAffine() will not return the updated values until the hardware configuration has been reloaded

marktsuchida, 2020-02-12:
I agree with the first 2 proposed improvements.

(Do you not see the pixel size take effect as soon as you exit the pixel size config, even if not saved?)

Regarding the Hardware Configuration Wizard, do you see this happen even if you don't actually make any changes?

nanthony21, 2020-02-12:
@marktsuchida . On a second look I am now seeing changes to the affine transform take effect immediately. I'm not sure what I must have done wrong earlier. I still think it would make sense to suggest that the user save changes when they close the pixel size calibration window.

Regarding the Hardware Configuration Wizard, yes. If I open a configuration and click "next" through the whole thing without making any changes, then at the end it will ask if I want to overwrite. The newly saved version will retain the pixel size but will have lost all references to the affine transform.

When I then reload the configuration micro-manager will not infer a correct affine transform from the pixel size information. The affine transform will be set to a default of [[1,0],[0,1]]

nicost, 2020-02-12:
Working on bullet point 1 (by doing heuristics and throwing up a dialog if it appears the user did not push Calculate when they should have.

Bullet point 2: Just as with changes to the config groups, MM should know that the config file was changed and prompt to safe it on exiting or changing config files.  If this is not the case, let's move this to its own issue.

Bullet point 3.  Works for me.  Start with 1-0-0-1 transform, enter pixel size of 0.25, click calculate (see number change), press Reset, and it works.

Bullet point 4.  I have been tracking a problems caused by lost affine transforms.  Looks like you found how that happened!  Let's move this into its own issue.

nicost, 2020-02-12:
Cross posted with Mark.  Split our bullet point 4 to #699.   I should have something for -1 shortly, and will investigate -2

nicost, 2020-02-12:
Just checked the remaining issue for Bullet point 2.  Whenever I make any change in pixel size configuration, the Save button in the top right side of the main window becomes active, and I am prompted to save the config file if I do not do so before exiting MM.  That is the intended behavior, and much less trouble some than prompting to save after every change to pixel size.  Is this not what you experience?

nicost, 2020-02-12:
@nanthony21 please move discussion of HCW to #699 (b.t.w., there never should be an unintended automatic editing of the affine transform)

As for saving, it would be highly annoying to be prompted to save the config for every change one makes (especially when setting up a system).  The current approach is fine (it should prompt you when quitting MM if you made any changes to the configuration), I am just wondering if you are running into a bug of sorts.

nicost, 2020-02-12:
PR #700 addresses bullet point 1.  BP 4 was moved to #699, and I believe that BP 2 and 3 were addressed (still unsure if @nanthony21 was never prompted to save the configuration or expected it to happen immediately, but did not realize that this was postponed till the end by design).  Closing for now, may be best to open individual issues if things persist.

closed 2020-02-12 by nicost

nanthony21, 2020-02-12:
Wow, you work fast! Thanks so much for getting these solved.

Regarding the question of asking the user to save after making changes to pixel calibration, I am seeing the expected behavior. While the current behavior does make sense to me, I have seen first hand that it is confusing for the standard user who isn't aware of the underlying MMCore+MMStudio structure of the code and doesn't understand the distinction between a hardware configuration and the many other settings that MMStudio saves automatically.

Here's an example scenario I've seen happen that illustrates my case:

1. User opens Micro-Manager and performs a calibration of the pixel affine transform. However, unlike the HCW which explicitly asks to save the configuration, and unlike the configuration groups which are located right next to the "Save" button, the pixel calibration is in it's own window and the user doesn't notice the "Save" button light up. The user assumes that the calibration is just like the many user-profile settings of Micro-Manager which silently save and load without the user ever noticing.
2. User uses Micro-Manager for a few hours and everything works fine.
3. After several hours of not having thought about pixel calibration, the user closes Micro-Manager and is asked to save the configuration. But the user thinks, "I haven't even opened the HCW. Everything with the configuration is already working fine." So to be safe they click "No".

However, if the user was asked to save the configuration file immediately after closing the pixel calibration dialog then it would be obvious to them that the need to save is related to the calibration.

nicost, 2020-02-13:
Good points!  The exact same thing goes for the channel groups and presets.  I really don't like prompting to save for every change, since that would be highly disturbing when setting up a new system (I have several with more than 20 pixel size configurations, would be a nightmare to be prompted for each and have to confirm that I want to over-write the existing config file).  I don't think that auto-saving the config file is a good idea either.

Suggestions for improved behavior?

nanthony21, 2020-02-13:
@nicost My feeling is that because the configuration groups are right next the "Save" button it is not quite as confusing, the user is much more likely to notice the button become enabled.

Since the pixel calibrations are set in their own dedicated dialog, I would think it would make sense to use the closing of that dialog as an opportunity to ask the user if they want to save. The user probably won't be closing and then reopening the dialog for every little change they make so it wouldn't be overly annoying.

nicost, 2020-02-13:
Makes sense, and I implemented it.  The only funny thing is that when the user says No, i.e. does not save the Pixel Size changes, they will still be in effect, and she/he will be prompted again on exit from the application.  That could be a bit confusing, but I'll take that for now.  Implemented in PR#703, should be in tonight's build.