micro-manager/micro-manager #698: Keyboard commands no longer move the stage in mm2-gamma

nicost, 2020-02-12:
The XYZListener code is still present in org.micromanager.internal.navigation, but it is no longer connected.  Keyboard control of stage movement is clearly desirable, adn we should re-implement this before 2.0-release.

There are various options how to go about this:
- We could use the same approach used in the UIMovesStageManager and ZWheelListeners, i.e. create events that originate from the Dataviewer and that are consumed by the XYZKeyListener.  Implementing this is probably straight forward, although some thought should be given to keyboard listener that ImageJ already attaches to the canvas (which should be disabled when we want to use the arrow keys).
- Alternatively, we can tie into HotKey mechanism (shortcuts), that uses MMKeyDispatcher to dispatch keys before they are seen elsewhere, I think that I favor this approach for simplicity.
- Lastly, it would be nice to somehow tie this in with the stage control plugin, but I am not exactly sure what that should look like.

julou, 2020-03-01:
Since I rose this in the first place, I feel that I should comment (to the extent that I understand what the discussion is about ;)

What made this particularly useful in 1.4 was combining it with modifier keys (mostly alt and shift for half and one field of view respectively), cf https://micro-manager.org/wiki/Micro-Manager_User%27s_Guide#Interactive_Stage_Movements

About UI/UX, I think it could be nice to have this coupled to the stage control plugin (but certainly not a requirement!). In particular, there are 3 types of displacement in this plugin and I think that the keyboard control should somehow mirror this (e.g. no modifier, alt and shift). As the plugin allows to specify the distance for each type of displacement, one possibly interesting option would be to have the keyboard control be shortcut for the plugin buttons (i.e. no modifier could be any displacement instead of 1 or 10 pixel and specified by what is set in the plugin window).

About which event listener to use, I have (of course) no opinion. I just want to mention that as far as I know current shortcuts don't allow for modifier keys and I believe those to be critical in this case… Hopefully those can be caught by MMKeyDispatcher and are simply not exposed in the shortcuts pref window.

julou, 2020-03-12:
⬆️  :)

nicost, 2020-04-03:
Addressed in PR #741.  @julou : this should work quite nicely once the PR has been merged in.

julou, 2020-04-03:
Awesome, I'll test it and report as soon as it makes it in the nightly build (provided I can still access the lab by then…)

nicost, 2020-04-03:
PR has been merged. I'll close this ticket, but feel free to open a new one discuss issues with the new code.

closed 2020-04-03 by nicost