micro-manager/micro-manager #708: MMClearVolume lose com.jogamp to compile in Linux

JiangXL, 2020-02-15:
I try to compile MicroManager(master: commit 68f3383) on Debian 9, then
``` bash
MMClearVolume.compile:
 [mm-javac] Compiling 14 source files to /home/qblab/Bin/mm-sh/micro-manager/plugins/build/MMClearVolume
 [mm-javac] warning: Supported source version 'RELEASE_6' from annotation processor 'com.jogamp.gluegen.structgen.CStructAnnotationProcessor' less than -source '1.8'
 [mm-javac] warning: No processor claimed any of these annotations: org.scijava.plugin.Plugin,com.google.common.eventbus.Subscribe
 [mm-javac] /home/qblab/Bin/mm-sh/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/CVViewer.java:24: error: package com.jogamp.newt.awt does not exist
 [mm-javac] import com.jogamp.newt.awt.NewtCanvasAWT;
 [mm-javac]                           ^
 [mm-javac] /home/qblab/Bin/mm-sh/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/CVViewer.java:28: error: package com.jogamp.newt does not exist
 [mm-javac] import com.jogamp.newt.NewtFactory;
 [mm-javac]                       ^
 [mm-javac] /home/qblab/Bin/mm-sh/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVVideoRecorder.java:5: error: package com.jogamp.opengl does not exist
 [mm-javac] import com.jogamp.opengl.GL;
 [mm-javac]                         ^
 [mm-javac] /home/qblab/Bin/mm-sh/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVVideoRecorder.java:6: error: package com.jogamp.opengl does not exist
 [mm-javac] import com.jogamp.opengl.GLAutoDrawable;
 [mm-javac]                         ^
 [mm-javac] /home/qblab/Bin/mm-sh/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVVideoRecorder.java:23: error: cannot find symbol
 [mm-javac]    public boolean screenshot(GLAutoDrawable pGLAutoDrawable,
 [mm-javac]                              ^
 [mm-javac]   symbol:   class GLAutoDrawable
 [mm-javac]   locati on: class CVVideoRecorder
 [mm-javac] /home/qblab/Bin/mm-sh/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVSnapshot.java:5: error: package com.jogamp.opengl does not exist
 [mm-javac] import com.jogamp.opengl.GL;
 [mm-javac]                         ^
 [mm-javac] /home/qblab/Bin/mm-sh/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVSnapshot.java:25: error: cannot find symbol
 [mm-javac]    public boolean screenshot(GLAutoDrawable pGLAutoDrawable,
 [mm-javac]                              ^
 [mm-javac]   symbol:   class GLAutoDrawable
 [mm-javac]   location: class CVSnapshot
 [mm-javac] 7 errors
 [mm-javac] 2 warnings
BUILD FAILED
```

So I install `libjogl2-java` using apt. Becease orignal build.xml seem only due with Windows and I no sure how to edit on existed item, I just fill `plugins/MMClearVolumePlugin/build.xml` with
``` xml
<project name="MMClearVolume" basedir="." default="jar">
        <import file="../javapluginbuild.xml"/>
        <path id="plugin.classpath">
                <pathelement location="/usr/share/java/jogl2.jar"/>
        </path>

</project>
```
Finally, MicroManager is built successfully. I think a better idea is required to fix linux compatibility.

i-mtz, 2020-02-20:
Thanks for the hint @JiangXL downloading the jar and configuring the path accordingly made the build to succeed.

JiangXL, 2021-08-04:
For the record in opensuse:
```bash
sudo zypper in jogl2
```
```xml
<pathelement location="/usr/lib64/java/jogl2.jar"/>
```

marktsuchida, 2022-03-21:
I believe this issue has been addressed by recent fixes and updated instructions. Please open a new issue with up-to-date details if not.

closed 2022-03-21 by marktsuchida