micro-manager/micro-manager #722: ubuntu 18 LTS build fails

dschneiderch, 2020-03-11:
Hi,
I am trying to install mm2 on ubuntu. it keeps failing during compilation. I have tried several fixes posted here:

1. using build scripts from @hadim https://github.com/hadim/mm_scripts - I tried this with and without older boost.
2. manually installed jogamp to get past error for Volume https://github.com/micro-manager/micro-manager/issues/708 but i still see an associated error for gluegen
```
[ivy:resolve] :: problems summary ::
[ivy:resolve] :::: WARNINGS
[ivy:resolve] 		module not found: org.jogamp.gluegen#gluegen-rt-local;2.3.2
```
but this should arready be installed:
```
couguser@cppc2:~$ sudo apt-get install libgluegen2-jni libgluegen2-rt-java libgluegen2-build-java libgluegen2-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libgluegen2-build-java is already the newest version (2.3.2-7~18.04).
libgluegen2-doc is already the newest version (2.3.2-7~18.04).
libgluegen2-jni is already the newest version (2.3.2-7~18.04).
libgluegen2-rt-java is already the newest version (2.3.2-7~18.04).
The following package was automatically installed and is no longer required:
  libnih-dbus1
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
```
[ivy:resolve] 		::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve] 		::          UNRESOLVED DEPENDENCIES         ::
[ivy:resolve] 		::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-local;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-main-local;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-natives-macosx-universal;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-natives-windows-amd64;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-natives-windows-i586;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.jogl#jogl-all-local;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.jogl#jogl-all-main-local;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.jogl#jogl-all-natives-macosx-universal;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.jogl#jogl-all-natives-windows-amd64;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.jogl#jogl-all-natives-windows-i586;2.3.2: not found
[ivy:resolve] 		:: #DT1.2;: not found
[ivy:resolve] 		::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve] :::: ERRORS
[ivy:resolve] 	SERVER ERROR: HTTPS Required url=http://repo1.maven.org/maven2/org/jogamp/gluegen/gluegen-rt-local/2.3.2/gluegen-rt-local-2.3.2.pom

... many more https errors here

```

compilation doesn't fail until OxxiusCombiner
3. now i'm stuck with ` InfoMessage4 << "test" << com;` during the OxxiusCombiner. I didnt understand how it was resolved in https://github.com/micro-manager/micro-manager/issues/681

```
make[2]: Leaving directory '/home/couguser/Applications/mm_scripts/micro-manager/DeviceAdapters/Oxxius'
Making all in OxxiusCombiner
make[2]: Entering directory '/home/couguser/Applications/mm_scripts/micro-manager/DeviceAdapters/OxxiusCombiner'
/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DPACKAGE_NAME=\"Micro-Manager\" -DPACKAGE_TARNAME=\"micro-manager\" -DPACKAGE_VERSION=\"1.4\" -DPACKAGE_STRING=\"Micro-Manager\ 1.4\" -DPACKAGE_BUGREPORT=\"info@micro-manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\" -DVERSION=\"1.4\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_BOOST=/\*\*/ -DHAVE_BOOST_THREAD=/\*\*/ -DHAVE_BOOST_ASIO=/\*\*/ -DHAVE_BOOST_SYSTEM=/\*\*/ -DHAVE_BOOST_CHRONO=/\*\*/ -DHAVE_BOOST_DATE_TIME=/\*\*/ -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1 -DSTDC_HEADERS=1 -DHAVE_MEMSET=1 -I.    -I/home/couguser/Applications/mm_scripts/micro-manager/DeviceAdapters/../MMDevice -pthread -I/home/couguser/Applications/mm_scripts/3rdpartypublic/boost-linux-x86_64/include -g -O2 -MT Oxxius_combiner.lo -MD -MP -MF .deps/Oxxius_combiner.Tpo -c -o Oxxius_combiner.lo Oxxius_combiner.cpp
libtool: compile:  g++ -DPACKAGE_NAME=\"Micro-Manager\" -DPACKAGE_TARNAME=\"micro-manager\" -DPACKAGE_VERSION=\"1.4\" "-DPACKAGE_STRING=\"Micro-Manager 1.4\"" -DPACKAGE_BUGREPORT=\"info@micro-manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\" -DVERSION=\"1.4\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" "-DHAVE_BOOST=/**/" "-DHAVE_BOOST_THREAD=/**/" "-DHAVE_BOOST_ASIO=/**/" "-DHAVE_BOOST_SYSTEM=/**/" "-DHAVE_BOOST_CHRONO=/**/" "-DHAVE_BOOST_DATE_TIME=/**/" -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1 -DSTDC_HEADERS=1 -DHAVE_MEMSET=1 -I. -I/home/couguser/Applications/mm_scripts/micro-manager/DeviceAdapters/../MMDevice -pthread -I/home/couguser/Applications/mm_scripts/3rdpartypublic/boost-linux-x86_64/include -g -O2 -MT Oxxius_combiner.lo -MD -MP -MF .deps/Oxxius_combiner.Tpo -c Oxxius_combiner.cpp  -fPIC -DPIC -o .libs/Oxxius_combiner.o
In file included from ../../MMDevice/MMDevice.h:49:0,
                 from Oxxius_combiner.h:15,
                 from Oxxius_combiner.cpp:14:
../../MMDevice/ImageMetadata.h:308:58: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
    MetadataSingleTag GetSingleTag(const char* key) const throw (MetadataKeyError)
                                                          ^~~~~
../../MMDevice/ImageMetadata.h:315:56: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
    MetadataArrayTag GetArrayTag(const char* key) const throw (MetadataKeyError)
                                                        ^~~~~
Oxxius_combiner.cpp: In member function ‘virtual int OxxiusCombinerHub::DetectInstalledDevices()’:
Oxxius_combiner.cpp:329:27: error: no match for ‘operator<<’ (operand types are ‘std::basic_ostream<char>’ and ‘std::ostringstream {aka std::__cxx11::basic_ostringstream<char>}’)
    InfoMessage4 << "test" << com;
    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
[... 374 lines cut ...]
## Micro-Manager configuration ##
## --------------------------- ##

   JAVA_HOME    = /usr/lib/jvm/java-11-openjdk-amd64
   JNI_CPPFLAGS = -I/usr/lib/jvm/java-11-openjdk-amd64/include -I/usr/lib/jvm/java-11-openjdk-amd64/include/linux
   JAVA         = /usr/lib/jvm/java-11-openjdk-amd64/bin/java
   JAVACFLAGS   = -Xlint:all,-path,-serial
   ANT          = JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64 /usr/bin/ant

   Build MMCoreJ (MMCore Java wrapper):    yes 
   Build MMCorePy (MMCore Python wrapper): no (Python not available)
   Build Java application (MMStudio):      yes 
   Build device adapters:                  yes
   Build closed-source device adapters:    no

   Install MMStudio as ImageJ plugin:      yes
   Target ImageJ directory:                /home/couguser/Applications/mm_scripts/Fiji.app
   ij.jar:                                 /home/couguser/Applications/mm_scripts/Fiji.app/jars/ij-1.52p.jar
   Install dependency JARs:                yes

First type 'make fetchdeps' once to download Java build dependencies.
Type 'make' to build.
Then type 'make install' to install.

```

nicost, 2020-03-12:
The https errors suggest (but I am not certain) that you are working of an older version (or svn-mirror) of the code.  Are you building off of the master branch?

Do you really need the Oxxius_Combiner?  You can delete the line with OxxiusCombiner in Makefile.am, and it should no longer be build.  If nobody on Ubuntu needs the OxxiusCOmbiner, I guess that we can figure out how to automate that.

dschneiderch, 2020-03-12:
@nicost  I am working off the master branch. it almost seems like that http: request is coming from something that was downloaded from ivy project?

dschneiderch, 2020-03-12:
@nicost Those http: are coming from `make fetchdeps`
```
mm.fetchdeps.download-ivy:
    [mkdir] Created dir: /home/couguser/Documents/mm_scripts/micro-manager/dependencies/ivy
      [get] Getting: https://repo1.maven.org/maven2/org/apache/ivy/ivy/2.3.0/ivy-2.3.0.jar
      [get] To: /home/couguser/Documents/mm_scripts/micro-manager/dependencies/ivy/ivy-2.3.0.jar

mm.fetchdeps.resolve:
[ivy:configure] :: Apache Ivy 2.3.0 - 20130110142753 :: http://ant.apache.org/ivy/ ::
[ivy:configure] :: loading settings :: file = /home/couguser/Documents/mm_scripts/micro-manager/buildscripts/ivysettings.xml
[ivy:resolve] :: resolving dependencies :: org.micromanager#micromanager;working@cppc2
[ivy:resolve] 	confs: [build, test, compile, optional, imagej, runtime]
[ivy:resolve] 	found ant-contrib#ant-contrib;1.0b3 in imagej-net
[ivy:resolve] 	found junit#junit;4.11 in imagej-net
[ivy:resolve] 	found org.hamcrest#hamcrest-core;1.3 in imagej-net
[ivy:resolve] 	found org.msgpack#msgpack;0.6.12 in imagej-net
[ivy:resolve] 	found com.googlecode.json-simple#json-simple;1.1.1 in imagej-net
[ivy:resolve] 	found org.javassist#javassist;3.18.1-GA in imagej-net
[ivy:resolve] 	found com.fifesoft#rsyntaxtextarea;2.6.1 in imagej-net
[ivy:resolve] 	found com.google.code.gson#gson;2.2.4 in imagej-net
[ivy:resolve] 	found com.google.guava#guava;17.0 in imagej-net
[ivy:resolve] 	found com.google.protobuf#protobuf-java;2.6.1 in imagej-net
[ivy:resolve] 	found com.miglayout#miglayout-swing;4.2 in imagej-net
[ivy:resolve] 	found com.miglayout#miglayout-core;4.2 in imagej-net
[ivy:resolve] 	found net.imglib2#imglib2;5.6.3 in imagej-net
[ivy:resolve] 	found ome#formats-api;5.1.1 in imagej-net
[ivy:resolve] 	found ome#formats-common;5.1.1 in imagej-net
[ivy:resolve] 	found com.esotericsoftware.kryo#kryo;2.24.0 in imagej-net
[ivy:resolve] 	found com.esotericsoftware.minlog#minlog;1.2 in imagej-net
[ivy:resolve] 	found org.objenesis#objenesis;2.1 in imagej-net
[ivy:resolve] 	found joda-time#joda-time;2.2 in imagej-net
[ivy:resolve] 	found org.slf4j#slf4j-api;1.7.6 in imagej-net
[ivy:resolve] 	found ch.qos.logback#logback-core;1.1.1 in imagej-net
[ivy:resolve] 	found ch.qos.logback#logback-classic;1.1.1 in imagej-net
[ivy:resolve] 	found xalan#serializer;2.7.1 in imagej-net
[ivy:resolve] 	found xml-apis#xml-apis;1.3.04 in imagej-net
[ivy:resolve] 	found xalan#xalan;2.7.1 in imagej-net
[ivy:resolve] 	found ome#ome-xml;5.1.1 in imagej-net
[ivy:resolve] 	found org.apache-extras.beanshell#bsh;2.0b6 in imagej-net
[ivy:resolve] 	found org.apache.commons#commons-lang3;3.5 in imagej-net
[ivy:resolve] 	found org.apache.commons#commons-math;2.2 in imagej-net
[ivy:resolve] 	found org.apache.commons#commons-math3;3.6.1 in imagej-net
[ivy:resolve] 	found org.clojure#clojure;1.3.0 in imagej-net
[ivy:resolve] 	found org.clojure#core.cache;0.6.2 in imagej-net
[ivy:resolve] 	found org.clojure#core.memoize;0.5.2 in imagej-net
[ivy:resolve] 	found org.clojure#data.json;0.1.1 in imagej-net
[ivy:resolve] 	found org.jfree#jcommon;1.0.24 in imagej-net
[ivy:resolve] 	found org.jfree#jfreechart;1.5.0 in imagej-net
[ivy:resolve] 	found org.scijava#scijava-common;2.77.0 in imagej-net
[ivy:resolve] 	found org.scijava#parsington;1.0.3 in imagej-net
[ivy:resolve] 	found org.bushe#eventbus;1.4 in imagej-net
[ivy:resolve] 	found org.zeromq#jeromq;0.5.1 in imagej-net
[ivy:resolve] 	found eu.neilalexander#jnacl;1.0.0 in imagej-net
[ivy:resolve] 	found io.scif#scifio;0.37.3 in imagej-net
[ivy:resolve] 	found io.scif#scifio-jai-imageio;1.1.1 in imagej-net
[ivy:resolve] 	found net.imagej#imagej-common;0.26.1 in imagej-net
[ivy:resolve] 	found net.imglib2#imglib2-roi;0.5.2 in imagej-net
[ivy:resolve] 	found net.imglib2#imglib2-realtransform;2.0.0 in imagej-net
[ivy:resolve] 	found gov.nist.math#jama;1.0.3 in imagej-net
[ivy:resolve] 	found jitk#jitk-tps;3.0.0 in imagej-net
[ivy:resolve] 	found log4j#log4j;1.2.17 in imagej-net
[ivy:resolve] 	found net.sf.trove4j#trove4j;3.0.3 in imagej-net
[ivy:resolve] 	found edu.ucar#udunits;4.3.18 in imagej-net
[ivy:resolve] 	found net.imglib2#imglib2-cache;1.0.0-beta-9 in imagej-net
[ivy:resolve] 	found com.github.ben-manes.caffeine#caffeine;2.4.0 in imagej-net
[ivy:resolve] 	found net.clearvolume#cleargl;2.2.6 in imagej-net
[ivy:resolve] 	found org.jogamp.gluegen#gluegen-rt-main;2.3.2 in imagej-net
[ivy:resolve] 	found org.jogamp.gluegen#gluegen-rt;2.3.2 in imagej-net
[ivy:resolve] 	found org.jogamp.jogl#jogl-all-main;2.3.2 in imagej-net
[ivy:resolve] 	found org.jogamp.jogl#jogl-all;2.3.2 in imagej-net
[... 517 lines cut ...]
[ivy:resolve] 	  -- artifact #DT1.2;!DT1.2.jar:
[ivy:resolve] 	  https://dl.bintray.com/haesleinhuepf/clij//DT1.2//DT1.2-.jar
[ivy:resolve] 		::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve] 		::          UNRESOLVED DEPENDENCIES         ::
[ivy:resolve] 		::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-local;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-main-local;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-natives-macosx-universal;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-natives-windows-amd64;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.gluegen#gluegen-rt-natives-windows-i586;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.jogl#jogl-all-local;2.3.2: not found
[ivy:resolve] 		:: org.jogamp.jogl#jogl-all-main-local;2.3.2: not found
[... cut ...]
[ivy:resolve] 	SERVER ERROR: HTTPS Required url=http://repo1.maven.org/maven2/org/jogamp/gluegen/gluegen-rt-main-local/2.3.2/gluegen-rt-main-local-2.3.2.jar
[ivy:resolve] 	SERVER ERROR: HTTPS Required url=http://repo1.maven.org/maven2/org/jogamp/gluegen/gluegen-rt-natives-macosx-universal/2.3.2/gluegen-rt-natives-macosx-universal-2.3.2.pom
[ivy:resolve] 	SERVER ERROR: HTTPS Required url=http://repo1.maven.org/maven2/org/jogamp/gluegen/gluegen-rt-natives-macosx-universal/2.3.2/gluegen-rt-natives-macosx-universal-2.3.2.jar
```

dschneiderch, 2020-03-12:
oddly, it says 
```
[ivy:resolve] 	found org.jogamp.gluegen#gluegen-rt-main;2.3.2 in imagej-net
[ivy:resolve] 	found org.jogamp.gluegen#gluegen-rt;2.3.2 in imagej-net
[ivy:resolve] 	found org.jogamp.jogl#jogl-all-main;2.3.2 in imagej-net
[ivy:resolve] 	found org.jogamp.jogl#jogl-all;2.3.2 in imagej-net
```

but then later says it can't find the same ones. I am now trying my branch that incorporates changes proposed by @triccyx 
https://github.com/CougPhenomics/micro-manager/tree/ubuntu1804

I don't know if the above is affecting later but now I"m stuck on ClearVolume again
```
MMClearVolume.compile:
    [mkdir] Created dir: /home/couguser/Documents/mm_scripts/micro-manager/plugins/build/MMClearVolume
 [mm-javac] Compiling 14 source files to /home/couguser/Documents/mm_scripts/micro-manager/plugins/build/MMClearVolume
 [mm-javac] warning: [options] bootstrap class path not set in conjunction with -source 8
 [mm-javac] warning: Supported source version 'RELEASE_6' from annotation processor 'com.jogamp.gluegen.structgen.CStructAnnotationProcessor' less than -source '8'
 [mm-javac] warning: No processor claimed any of these annotations: org.scijava.plugin.Plugin,com.google.common.eventbus.Subscribe
 [mm-javac] /home/couguser/Documents/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/CVViewer.java:24: error: package com.jogamp.newt.awt does not exist
 [mm-javac] import com.jogamp.newt.awt.NewtCanvasAWT;
 [mm-javac]                           ^
 [mm-javac] /home/couguser/Documents/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/CVViewer.java:28: error: package com.jogamp.newt does not exist
 [mm-javac] import com.jogamp.newt.NewtFactory;
 [mm-javac]                       ^
 [mm-javac] /home/couguser/Documents/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVVideoRecorder.java:5: error: package com.jogamp.opengl does not exist
 [mm-javac] import com.jogamp.opengl.GL;
 [mm-javac]                         ^
 [mm-javac] /home/couguser/Documents/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVVideoRecorder.java:6: error: package com.jogamp.opengl does not exist
 [mm-javac] import com.jogamp.opengl.GLAutoDrawable;
 [mm-javac]                         ^
 [mm-javac] /home/couguser/Documents/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVVideoRecorder.java:23: error: cannot find symbol
 [mm-javac]    public boolean screenshot(GLAutoDrawable pGLAutoDrawable,
 [mm-javac]                              ^
 [mm-javac]   symbol:   class GLAutoDrawable
 [mm-javac]   location: class CVVideoRecorder
 [mm-javac] /home/couguser/Documents/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVSnapshot.java:5: error: package com.jogamp.opengl does not exist
 [mm-javac] import com.jogamp.opengl.GL;
 [mm-javac]                         ^
 [mm-javac] /home/couguser/Documents/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVSnapshot.java:25: error: cannot find symbol
 [mm-javac]    public boolean screenshot(GLAutoDrawable pGLAutoDrawable,
 [mm-javac]                              ^
 [mm-javac]   symbol:   class GLAutoDrawable
 [mm-javac]   location: class CVSnapshot
 [mm-javac] 7 errors
 [mm-javac] 3 warnings

BUILD FAILED
/home/couguser/Documents/mm_scripts/micro-manager/plugins/build.xml:23: The following error occurred while executing this line:
/home/couguser/Documents/mm_scripts/micro-manager/plugins/build.xml:7: The following error occurred while executing this line:
/home/couguser/Documents/mm_scripts/micro-manager/buildscripts/javabuild.xml:112: Compile failed; see the compiler error output for details.

Total time: 13 seconds
Makefile:579: recipe for target 'plugins.stamp' failed
make[1]: *** [plugins.stamp] Error 1
make[1]: Leaving directory '/home/couguser/Documents/mm_scripts/micro-manager/plugins'
Makefile:521: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
```

triccyx, 2020-03-12:
This solve the problem:
https://github.com/micro-manager/micro-manager/issues/708

dschneiderch, 2020-03-12:
yah! It finished building using this fork https://github.com/CougPhenomics/micro-manager/tree/ubuntu1804
I basically took the changes from @triccyx and included the fix from #708 .

But, MM does not run because of MMCore_wrap error. it cant find a boost file.

```
couguser@cppc2:~/Applications$ ./Fiji.app/mmimagej 
Gtk-Message: 11:35:23.803: Failed to load module "canberra-gtk-module"
Load MMCoreJ_wrap: Start loading...
Load MMCoreJ_wrap: Will search in hard-coded paths:
Load MMCoreJ_wrap:   /home/couguser/Documents/mm_scripts/Fiji.app/plugins/Micro-Manager
Load MMCoreJ_wrap:   /home/couguser/Documents/mm_scripts/Fiji.app/plugins
Load MMCoreJ_wrap:   /home/couguser/Documents/mm_scripts/Fiji.app/plugins/mm/linux64
Load MMCoreJ_wrap:   /home/couguser/Documents/mm_scripts/Fiji.app
Load MMCoreJ_wrap:   /home/couguser/Documents/mm_scripts/Fiji.app/mm/linux64
Load MMCoreJ_wrap:   /usr/local/lib/micro-manager
Load MMCoreJ_wrap: Skipping nonexistent candidate: /home/couguser/Documents/mm_scripts/Fiji.app/plugins/Micro-Manager/libMMCoreJ_wrap.so
Load MMCoreJ_wrap: Skipping nonexistent candidate: /home/couguser/Documents/mm_scripts/Fiji.app/plugins/libMMCoreJ_wrap.so
Load MMCoreJ_wrap: Skipping nonexistent candidate: /home/couguser/Documents/mm_scripts/Fiji.app/plugins/mm/linux64/libMMCoreJ_wrap.so
Load MMCoreJ_wrap: Try loading: /home/couguser/Documents/mm_scripts/Fiji.app/libMMCoreJ_wrap.so
Failed to load the MMCoreJ_wrap native library
java.lang.UnsatisfiedLinkError: /home/couguser/Documents/mm_scripts/Fiji.app/libMMCoreJ_wrap.so: libboost_system.so.1.57.0: cannot open shared object file: No such file or directory in Thread[AWT-EventQueue-0,6,main]
  at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
  at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2430)
  at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2487)
  at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2684)
  at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2617)
  at java.base/java.lang.Runtime.load0(Runtime.java:767)
  at java.base/java.lang.System.load(System.java:1834)
  at mmcorej.MMCoreJJNI.loadNamedNativeLibrary(MMCoreJJNI.java:119)
  at mmcorej.MMCoreJJNI.loadNativeLibrary(MMCoreJJNI.java:145)
  at mmcorej.MMCoreJJNI.loadFromHardCodedPaths(MMCoreJJNI.java:205)
  at mmcorej.MMCoreJJNI.<clinit>(MMCoreJJNI.java:223)
  at mmcorej.CMMCore.<init>(CMMCore.java:304)
  at org.micromanager.internal.MMStudio.<init>(MMStudio.java:269)
  at MMStudioPlugin$1.run(MMStudioPlugin.java:81)
  at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
  at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
  at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
  at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
  at java.base/java.security.AccessController.doPrivileged(Native Method)
  at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
  at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
  at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
  at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
  at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
  at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
  at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
  at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

Searching for plugins in /home/couguser/Documents/mm_scripts/Fiji.app/mmplugins

java.lang.NullPointerException in Thread[AWT-EventQueue-0,6,main]
  at org.micromanager.internal.MMStudio.<init>(MMStudio.java:303)
  at MMStudioPlugin$1.run(MMStudioPlugin.java:81)
  at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
  at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
  at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
  at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
  at java.base/java.security.AccessController.doPrivileged(Native Method)
  at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
  at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
  at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
  at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
  at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
  at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
  at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
  at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

Found plugin org.micromanager.intelligentacquisition.IntelligentAcquisition@3fa229cf
[... 36 lines cut ...]
Found plugin org.micromanager.display.inspector.internal.panels.metadata.SummaryMetadataInspectorPanelPlugin@34fc191d
Found plugin org.micromanager.display.inspector.internal.panels.metadata.PlaneMetadataInspectorPanelPlugin@2825ba6e
[... cut ...]
Found plugin org.micromanager.quickaccess.internal.controls.RefreshButton@7e546cc3
Found plugin org.micromanager.quickaccess.internal.controls.SavedMDAButton@1e3e8670
Found plugin org.micromanager.quickaccess.internal.controls.ScriptButton@48403a61
Found plugin org.micromanager.quickaccess.internal.controls.ShutterControl@325687c4
Found plugin org.micromanager.quickaccess.internal.controls.SnapButton@5b5114d6
Found plugin org.micromanager.quickaccess.internal.controls.TextLabel@4734945d
Plugin loading took 2145ms
```

triccyx, 2020-03-12:
Have you tried:
```bash
sudo apt-get install libboost-all-dev
```

dschneiderch, 2020-03-12:
evidently yes:
 libboost-all-dev is already the newest version (1.65.1.0ubuntu1).

triccyx, 2020-03-12:
It seems that libMMCoreJ_wrap.so has been compiled against
libboost_system.so.1.57.0 but you have 1.65.1

Is it possible that you have ./configure before install libboost 1.65.1? If
so try:

```bash
make clean
./configure
sudo make install
```
otherways we need a Java man ...

dschneiderch, 2020-03-12:
i then tried to install with the latest ubuntu libboost instead of the older 1.57 from anaconda as in https://github.com/CougPhenomics/mm_scripts/blob/master/build-mm.sh
and got this error:
```
Bootstrapping complete; now you can run ./configure
If you would like to install Micro-Manager as an ImageJ plugin, run
  ./configure --enable-imagej-plugin=/path/to/ImageJ
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to print strings... printf
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
checking for clang++... no
checking for llvm-g++... no
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for gmock source in ./testing/gmock... no
checking for boostlib >= 1.48.0... yes
checking whether the Boost::Date_Time library is available... yes
configure: error: Could not find a version of the library!
```

dschneiderch, 2020-03-12:
yes, 1.57 was being used because others reported the newer version not working.    I just tried to run make clean inside my micro-manager folder but i get the same error that I posted above.

I deleted all the folders it had downloaded too and started over but it can't get past the boost version error.

dschneiderch, 2020-03-12:
i figured that out. the boost version was missing because I was still pointing to the custom boost folder during ./configure
I commented that out and it compiled and started up!
Here is the build script I used https://github.com/CougPhenomics/mm_scripts/blob/master/build-mm.sh

closed 2020-03-12 by dschneiderch