micro-manager/micro-manager #723: mm.data().convertTaggedImage() is slow for large images

nicost, 2020-03-12:
While investigating bottlenecks in the Micro-Manager data pipeline, I discovered that the function mm.data().convertTaggedImage() is rate limiting for large (2k x 2k x 2 byte) images.  The upside is that metadata apparently are not an issue, the downside that this most likely is caused by the conversion of a pixel array into a DirectByteBuffer.  The following script illustrates the issue.  This problem limits the speed with which data can be stored in RAM, hence is a serious limitation in performance.

```
/**
 * Testing which part is slow in handling images in Micro-Manager
 * Use the Democamera, 2048x2048, 16 bit, 5 msec exposure
 * Snap one image, then set the "FastImage" property to 1
 * 
 * 2000 images, so ideally this will take 10000 ms.
 * 
 * when not calling mmc.data().convertTaggedImage() we get:
 * mean: 10021 +- 7 (n=5)
 * 
 * Stage 2 of testing: converting the Tagged Image to a MM Image 
 * This takes considerable time:
 * 17688 (and lost images total 1767)
 * 17999 (2000) 17954 (2000) 17821 (2000) 17616 (2000) 17785 (2000)
 * mean: 17835 +- 151 (n=5)
 * 
 * With 512 x 512:
 * 10021 (2000) 10004  10023 10016 10020
 * mean: 10017 +- 8 (n=5)
 * 
 * 512 x 512 1 ms exposure
 * 2131 2303 2477 2366 2456
 * mean 2347 +- 139
 */

nrFrames = 2000;
exposureMs = mmc.getExposure();

builder = mm.data().getCoordsBuilder().z(0).channel(0).stagePosition(0);
coord = builder.build();

mmc.startSequenceAcquisition(nrFrames, 0, true);

int curFrame = 0;
start = System.currentTimeMillis();
while (mmc.getRemainingImageCount() > 0 || 
			mmc.isSequenceRunning(mmc.getCameraDevice())) {
   if (mmc.getRemainingImageCount() > 0) {
      tagged = mmc.popNextTaggedImage();
      // Convert to an Image at the desired timepoint.
      image = mm.data().convertTaggedImage(tagged,
         coord, null);
      curFrame++;
   }
   else {
      mmc.sleep(Math.min(.5 * exposureMs, 20));
   }
}

end = System.currentTimeMillis();

mmc.stopSequenceAcquisition();

duration = end - start;
mm.scripter().message("It took: " + duration + " milliseconds");  
mm.scripter().message("Stored " + curFrame + " images.");

```

marktsuchida, 2020-03-16:
Historically, I was told by Arthur, that the reason for using direct byte buffers was that they could be allocated faster than Java arrays. But it is totally possible that this has changed with Java and OS versions.

The only real solution I can think of is to avoid the extra copies entirely. Right now we copy from C array to Java array (in MMCoreJ), then back to direct byte buffer. Creating a direct byte buffer directly in MMCoreJ could replace two byte-swapping copies with one non-swapping copy. (Actually even that single copy can be eliminated, but that would require a total rewrite of MMCore.)

I don't think there is an easy short-term fix for this, but in the mid/long term we should make sure that the replacement for `TaggedImage` coming out of MMCoreJ uses a direct byte buffer from the start.

nicost, 2020-03-23:
@marktsuchida My memory is that Arthur found that saving files from Java arrays was not fast enough, but that saving from Direct ByteBuffers was.  However, allocation of DirectBYteBUffers was slow, which is why he made a small pool of pre-allocated ByteBuffers that was continuously re-used.  
It therefore seems that the way MM 1.4 handles memory is significantly more efficient than MM 2.0.  When data are kept in memory, the arrays in TaggedImage are not converted into something else, but used directly, conversion into DirectByteBuffers only happens when saving to file, but re-using the same buffers speeds things up considerably over continuously allocating (and presumably unnecessary zeroing by Java).  
I am not sure what strategy gives the least amount of work, but I as the MM2 Image class abstract pixel storage, it may be possible to change the implementation so that it can store pixels again as Java arrays (so that no conversion is needed when translating a TaggedImage), and provide an option to get pixels out as a Direct ByteBuffer (and implement a small number of DirectByteBuffers that can be re-used).  That would basically return to the strategy used in 1.4, but with better API, and we can do the mid/long term fix later. Thoughts?

nicost, 2020-03-25:
Changing data/internal/DefaultImage.java:DefaultImage(TaggedImage) from:

`rawPixels_ = DirectBuffers.bufferFromArray(tagged.pix);`

to 

`rawPixels_ = ByteBuffer.wrap((byte[]) tagged.pix);`
and
`rawPixels_ = ShortBuffer.wrap((short[]) tagged.pix);`

reduced the duration of the script to 10015 ms (i.e., the extra overhead disappeared.  Display of images is fine, as well as saving snapshots.  Will investigate the effect on saving fast time lapses to disk.

nicost, 2020-03-26:
Situation was even more dire.  Not only was DefaultImage allocating a new DirectBuffer for each image and copying the pixel into it, whenever pixels were requested through DefaultImage.getRawPixels(), they were copied back from the DirectBuffer into a Java array.  Luckily, it is quite simple to keep the original data, skipping Direct Buffer allocation and copying data twice through the JVM layer, which must improve things.  I started a branch (https://github.com/nicost/micro-manager/tree/fasterSaving) implementing these changes.

Regretfully, the code still can not write multi-pageTiff files fast enough to keep up with 2k x 2k x 2 byte images (generated using the fastcam option of the DemoCamera) at 100fps, which is my goal (knowing this was possible in the past).  Luckily, my laptop drive saves data at > 2 GB/s, so I have a decent test setup;).  Will dig in the TIFF saving code next. Also will need to check in on 3D Viewer, as that was using the direct buffers for transfer to GPU.

nicost, 2020-04-03:
All bottlenecks were addressed in PR#735.  See discussion there for full details.  Fast saving of 2k x 2k x 2 byte x 100fps is possible again.

closed 2020-04-03 by nicost