micro-manager/micro-manager #726: createSinglePlaneTIFFSeriesDatastore and createMultipageTIFFDatastore produce different metadata text files

ptbrown1729, 2020-03-17:
When using `createMultipageTIFFDatastore` the metadata.txt file that is produced omits a number of useful metadata fields that _are_ present when using `createSinglePlaneTIFFSeriesDatastore`. In particular, Width, Height, PixelType, Frame, FrameIndex, PositionIndex, Slice, SliceIndex, ChannelIndex. 

For example, running the following script,
```
import org.micromanager.data.Coordinates;

int slices = 6;
int tPoints = 4;
int channels = 3;

cb = Coordinates.builder();

cb.c(1).t(1).p(1).z(1);

//store = mm.data().createMultipageTIFFDatastore("C:\\Users\\shepherdlab\\Desktop\\test_multipagetiff", true, false);
store = mm.data().createSinglePlaneTIFFSeriesDatastore("C:\\Users\\shepherdlab\\Desktop\\test");

for (int z = 0; z < slices; z++) {
	for (int t = 0; t < tPoints; t++) {
		for (int c = 0; c < channels; c++) {
			cb.c(c).t(t).p(1).z(z);
			imgs = mm.live().snap(false);
			store.putImage(imgs.get(0).copyAtCoords(cb.build()));
		}
	}
}
```

Using `createMultipageTIFFDatastore` the first entry after "Summary" of my metadata.txt file looks like:

```
"FrameKey-0-0-0": {
  "UUID": "348e56cb-a42e-4730-8812-249ee0469b8b",
  "Camera": "HamamatsuHam_DCAM",
  "Binning": 1,
  "ROI": "0-0-2048-2048",
  "BitDepth": 16,
  "PixelSizeUm": 0.0645,
  "PixelSizeAffine": "1.0;0.0;0.0;0.0;1.0;0.0",
  "PositionName": "Default",
  "UserData": {...
},
  "FileName": "MMStack.ome.tif"
}
```

But using `createSinglePlaneTIFFSeriesDatastore` it looks like:
```
Coords-Default/img_channel000_position001_time000000000_z000.tif": {
  "Frame": 0,
  "FrameIndex": 0,
  "PositionIndex": 1,
  "Slice": 0,
  "SliceIndex": 0,
  "ChannelIndex": 0
},
"Metadata-Default/img_channel000_position001_time000000000_z000.tif": {
  "Width": 2048,
  "Height": 2048,
  "PixelType": "GRAY16",
  "Frame": 0,
  "FrameIndex": 0,
  "PositionIndex": 1,
  "Slice": 0,
  "SliceIndex": 0,
  "ChannelIndex": 0,
  "UUID": "57115cf7-fa39-406d-9912-ad49ee50f955",
  "Camera": "HamamatsuHam_DCAM",
  "Binning": 1,
  "ROI": "0-0-2048-2048",
  "BitDepth": 16,
  "PixelSizeUm": 0.0645,
  "PixelSizeAffine": "1.0;0.0;0.0;0.0;1.0;0.0",
  "PositionName": "Default",
  "UserData": {...
  },
  "FileName": "Default/img_channel000_position001_time000000000_z000.tif"
}
```

nicost, 2020-04-04:
Related or identical to #465

nicost, 2020-04-30:
Sorry to it took so long to look into this.  The code on lines 179 - 191 in StorageSinglePlaneTiffSeries (org.micromanager.data.internal) creates this metadata.  I could add similar code to the MultiPLageTIff storage (should probably be n FileSet.java in the function writeImage), but I am not sure what is most desirable.  

Adding this code will add some overhead, not much, but possibly a nuisance for everyone wanting high performance.   Also, currently, Width, Height, and PixelType will be identical for all images, so it would make more sense to store this in the Summary Metadata.  Also, it does not make sense to have both Frame and FrameIndex (same for the others).  org.micromanager.data.internal.NonPropertyMapJSONFormats suggests that Frame is preferred in the current code.  So, my preference is to add Frame, PositionIndex, Slice, and ChannelIndex (no idea why two have index and the other two have not) to the ImageMetadata, and Width, Height, and PIxelType to the SummaryMetadata, but I am very open to suggestions!  @marktsuchida your input will be appreciated!

marktsuchida, 2020-08-31:
I think the more fundamental issue is that the single-plane `metadata.txt` format got changed (see how it no longer uses `FrameKey-*` keys) but it appears that the multi-plane format's `*_metadata.txt` files never got the corresponding update. I also think the multi-plane version is missing some stuff that the older single-plane format had, in particular the mapping from coordinates to `FrameKey` numbers (I could be wrong). I don't feel like I can come to any conclusion about individual fields and where they should be recorded without some more study of the code and its history. I do intend to get to this.