micro-manager/micro-manager #738: Add option to the API to get the currently loaded config file

nicost, 2020-03-31:
The StageControl plugin needs MMStudio.getInstance() to get to getSysConfigFile().  It is reasonable to assume that other plugins/scripts will want access to that information as well.  It makes most sense to me to add that to the Core (which is the entity actually loading the config file, so it seems straight forward to store a string with the last loaded config file and return that in a function.

marktsuchida, 2020-04-02:
I think the API should be added to `Studio`.

Maybe it could be added to the Core, but that would only make sense if a lot more logic is moved to the Core. As it stands now, the Core is completely stateless in regard to the configuration file, and it is entirely MMStudio that keeps track of that state. There are advantages to this, too: there is all the stuff about "recently used" configs, including the per-profile last-used config. Certainly we don't want to move that into the Core, but it will get tricky if the Core has its own memory and MMStudio needs to second guess what's happening (or get callbacks?). (I won't even get into how the Hardware Configuration Wizard should handle config files -- it's not ideal how it works currently either.) To me it seems much more logical to keep all config file state in the same module, i.e., MMStudio. Also because MMStudio is likely to need its own state management of the current config anyway (unless we add quite an elaborate interface to the Core), and it is best to have fewer sources of the same truth.

Just to clarify, I'm not 100% opposed to adding this to the Core, I just don't want to see a half-way situation in which you end up needing to look at both the Core and the Studio to get a full picture.

~~What is the reason that you don't want `StageControlFrame` to have access to the Studio? It looks like it already caches `studio_` when constructed.~~ I see you are referring to the fact that the internal interface is needed (which could be just a cast, no need for static method).

nicost, 2020-04-02:
@marktsuchida Where in Studio shall we add it?  Browsing the API, the only place that makes sense is the top level Studio itself, which would be ugly since Studio has no other top level methods, only gives access to managers. Make a configuration manager instead?  That seems a heavy solution to a small problem.  On the other hand there are quite a few events that are related to configuration that are posted by Studio rather than a configuration manager.  Ideas?

> What is the reason that you don't want StageControlFrame to have access to the Studio? It looks like it already caches studio_ when constructed. I see you are referring to the fact that the internal interface is needed (which could be just a cast, no need for static method).

Correct.  I'll do the casting instead of the static method for now, but I like opening tickets for every issue we encounter where plugin code is using internal functions.  We could say that StageControl is not a plugin, but it surely started its life out as one, and I can very much see that other plugins will need access to this information.

marktsuchida, 2020-08-31:
I think creating a `HardwareConfigurationManager` is in fact the best way, even if (for now) it only has one API method. It seems like a good opportunity to factor out all the config-management code from `MMStudio`.