micro-manager/micro-manager #740: OlympusX83 error

triccyx, 2020-04-02:
System:
- Windows 10
- Visual Studio 10
- mm 2.0 last nightly build
- Olympus module mmgr_dal_OlympusIX83.dll and mmgr_dal_Olympus.dll
- MMCore.dll compiled with vc2010 SDK 2007

while I'm trying:

```c++
        CMMCore core;
	std::string info = core.getVersionInfo();
	std::cout << "Version:" << info;

	//log
	core.enableDebugLog(true);
	core.setPrimaryLogFile("../../log/log.log", true);

	//path
	std::vector<std::string> modulePaths;
	modulePaths.push_back("../../modules");
	core.setDeviceAdapterSearchPaths(modulePaths);
        core.unloadAllDevices();
	core.loadDevice("Olympus IX83", "OlympusIX83", "Olympus IX83");
	core.loadDevice("Light Path", "OlympusIX83", "Light Path");
	core.loadDevice("Dichroic 1", "OlympusIX83", "Dichroic 1");
	core.loadDevice("EpiShutter 1", "OlympusIX83", "EpiShutter 1");
	core.loadDevice("TransmittedIllumination 1", "OlympusIX83", "TransmittedIllumination 1");
	core.loadDevice("TransmittedIllumination 2", "OlympusIX83", "TransmittedIllumination 2");
	core.loadDevice("FocusDrive", "OlympusIX83", "FocusDrive");
	try
	{
		core.initializeAllDevices();
	}
	catch (const std::exception& e)
	{
		return;
	}
```

an exception is thrown in initializeAllDevices:

```
"Error in device \"Olympus IX83\": Exception: Cannot create child process at e:\\mm2gammagitbuild\\micro-manager\\secretdeviceadapters\\olympusix83\\Hub.h:130 (2147483647)"
```
 Any idea?
The OlympusX83 currently is answering on the serial using the PortManager application by Olympus.
In the case is it possible to have the source files for debugging?

nicost, 2020-04-02:
It would be really great if we can share the source code with you.  However, we have signed agreements with Olympus not to share our source code (as long as it pertains to Olympus equipment).  However, if you would get permission from Olympus to work on this code, then I would see no problem.

Nevertheless, your error message does lead me to the source of the error.  The code makes a call to the failing WIN32 call:
` if (!CreateProcessA(NULL, cmdline,
         NULL, NULL, TRUE, 0, NULL, NULL, &startupInfo, &procInfo)) {
            throw std::runtime_error("Cannot create child process");
      }
`
relevant code preceding this:
`      helperExe = ourDLLDir + "/OlympusIX3Control/OlympusIX3Control.exe";
      char cmdline[256];
      strncpy(cmdline, (helperExe + " computer").c_str(), 255);
`

If you have the OlympusIX83Control.exe in the right place, then my first suspicion would be a security issue (i.e. Windows 10 no longer allowing this kind of thing).

triccyx, 2020-04-02:
Ok, I understand OlympusIX83Control folder should be in the dll folder. This first error has gone.
(I have others)
Thanks!

closed 2020-04-02 by triccyx

nicost, 2020-04-03:
Strange this was not in the documentation.  I added it: https://micro-manager.org/wiki/OlympusIX83#Olympus_IX83_device_adapter.

Puccet, 2020-04-20:
Dear Mr. Stuurman,

I'm a colleague of Luca, we are in contact with Olympus Europe, could we have your email to get you in the loop since they're asking us some details about what kind of Agreement you currently have with Olympus and your contact reference details. Did you sign the agreement with Olympus USA or Japan?

We would be very grateful for your support since keep the system running on Linux for us is a major requirement.

Thanks in advance

Fabio Frumento

nicost, 2020-04-20:
Hi @Puccet , Yes, please do keep me in the loop!  Send a trouble report (Help > Report a problem) from Micro-Manager to get in touch.  I'll start looking for that agreement.  If it is not in my email archive, I will need to go into the lab for it, which may delay things;)

Wonderful that you are so motivated to fix this!  It would be really great for someone with that microscope to work on that code. B.t.w., I though that a windows driver was needed for this microscope.  How do you deal with that on linux?

Puccet, 2020-04-20:
Hi @nicost, Thank you very much, I'll write to the help shortly.

About the device driver on Linux we have to speak with Olympus as well, Linux has native firewire support and the microscope interface is just an emulated serial port (the protocoll is ASCII based), not sure if it presents itself exactly as serial profile or custom profile with some ioctl calls, we have a little budget also for the device driver development if things will go for the worst.