micro-manager/micro-manager #747: Java layer only propagates stage callbacks for the default stage

nicost, 2020-04-04:
CoreEventCallback has the following code:

> public void onStagePositionChanged(String deviceName, double pos) {
>       // TODO: this check should be in the core, not the java layer!
>       if (deviceName.equals(core_.getFocusDevice())) {

and callbacks for anything other the current focusdevice are discarded.  I have no idea why, and also do not think that this check should be in the core layer.  For instance, the StageControlFrame could use these callbacks to update the stage position indicators.  Same goes for the XYStage callback.

@marktsuchida OK to delete comment and check for current device?

marktsuchida, 2020-04-05:
I agree that it is not clear why only the current stage is relayed, but simply removing the check will break code that subscribes to this (well, at least the ones in `StaticInfo`). Should be ok if they are also updated.

(The check could be moved to `StaticInfo`, or `StaticInfo` could be extended to monitor all stages. The latter seems like a better direction in the long term (the notion of the "current" stage, shutter, etc., are great in the user interface but mostly just get in the way when coding), but it probably doesn't make sense right now -- `StaticInfo` itself looks like a workaround for other problems.)

nicost, 2020-04-05:
Funny thing was that StaticInfo already checked the stage name, so the rational for only relaying the current stage in CoreEventCallback is even more mysterious.  Removed in PR #753

closed 2020-04-05 by nicost