micro-manager/micro-manager #750: OlympusIX83 Objective error

triccyx, 2020-04-05:
System:

Windows 10
Visual Studio 10
mm 2.0 last nightly build
Olympus module mmgr_dal_OlympusIX83.dll and mmgr_dal_Olympus.dll
MMCore.dll compiled with vc2010 SDK 2007

while I'm trying:
```c++
        core.unloadAllDevices();
	core.loadDevice("Olympus IX83", "OlympusIX83", "Olympus IX83");
	core.loadDevice("Objective", "OlympusIX83", "Objective");
	core.loadDevice("Light Path", "OlympusIX83", "Light Path");
	core.loadDevice("Dichroic 1", "OlympusIX83", "Dichroic 1");
	core.loadDevice("EpiShutter 1", "OlympusIX83", "EpiShutter 1");
	core.loadDevice("TransmittedIllumination 1", "OlympusIX83", "TransmittedIllumination 1");
	core.loadDevice("TransmittedIllumination 2", "OlympusIX83", "TransmittedIllumination 2");
	core.loadDevice("FocusDrive", "OlympusIX83", "FocusDrive");
	try
	{
		core.initializeAllDevices();
	}
	catch (CMMError& e)
	{
		std::cout << "Error:" << e.what() << std::endl;
		core.unloadAllDevices();
		return;
	}
```
I have this exception if I have the Objective between the loadDevice, otherways all
gone well :

```c++
message_ = "Error in device \"Objective\": Exception: Expected an integer. Instead got OB X at e:\\mm2gammagitbuild\\micro-manager\\secretdeviceadapters\\olympusix83\\StateDevice.h:187 (2147483647)"
```
Any ideas? Thanks!

nicost, 2020-04-05:
Hard to say what is going wrong here without access to both code and hardware.  The line in question is:
> try {
            pProp->Set(controller_->GetIntState(settings_.commandPrefix) - settings_.offset);
         } CATCH_AND_RETURN_ERRORS

but that seems to work well for other Statedevices (such as your Dichroic 1), and presumably worked in the past as well, so this will be harder to untangle.  Did you ask Olympus for permission to work with the code? 
Can you generate a trouble report (easiest to do from the MM UI, Help > Report a problem, and go through the Hardware configuration wizard, alternatively, switch on debug logging in your script, and email me the corelog.

triccyx, 2020-04-05:
Hi!
` Did you ask Olympus for permission to work with the code?`
Yes, but at the moment I have not an answer.
`Can you generate a trouble report...`
Yes, I have sent it.  I have the error in my c++ code but there is the same error in HW configuration wizard from MM.

triccyx, 2020-04-06:
I have tried the objective with direct serial port command. Same response (OB X). So I suppose that the problem is in the OlympusIX83 HW. As now we are work remotely I will wait that someone can check the situation on-site.

I have a question about the relation between the MM OlympusIX83 module and the application 
OlympusIX3Control.exe.

closed 2020-04-14 by triccyx