micro-manager/micro-manager #758: Can (parts) of internal.utils become part of the API?

nicost, 2020-04-06:
While annotating plugins with warnings against using internal packages (issue #372), it became clear that the use of internal packages is actually advisable.  Most of the plugins use packages from internal.utils for good reason.  Not using them would result in much less uniform UI behavior (undesirable for users), and/or massive code duplication.  @marktsuchida  how do we proceed?  

The most widely used internal.utils classes are:
MMFrame
MMDialog
FileDialogs
GUIUtils
ProgressBar
NumberUtils
TextUtils
JavaUtils
MathFunctions
AffineUtils

marktsuchida, 2020-04-16:
I think many parts of these can indeed be turned into APIs. Some of the grouping may need revision, some methods are inappropriate, etc., and it's mostly about sorting out details.

For example, `NumberUtils` is confusingly named. It's about Core device property values, nothing to do with numbers in general. The `AffineUtils` methods for conversion to/from `mmcorej.DoubleVector` better belong together with these than they do with the mathematical methods on affine transforms (which don't depend on `mmcorej`).

`TextUtils` is a heterogeneous mix of two obsolete file utilities (see `Files.readAllBytes` et al.), one string truncation method which is indeed useful but having more to do with GUI than text, and some number formatting stuff that really should be defined at the point of usage (or merged into `NumberUtils.doubleToDisplayString()`?). They clearly don't belong together.

`JavaUtils` looks like a dumping ground for methods that people didn't know where to put, and some of them have implementations that are highly suspect.

And so on. In short, some of the functionality definitely should be in the API but they are not currently written in the form of a well organized API. I think some amount of "miscellaneous utilities" classes are okay for internal-only use, but in the API at least each class should have clearly delineated responsibilities.

`MMFrame` and `MMDialog` are a slightly different story and can hopefully be retired in favor of `WindowPositioning`, but the latter might need some more testing and has some TODO comments. `MMFrame` and `MMDialog` are broken when you already need a frame or dialog to inherit from something else (or when you are given an existing `JFrame` instance created by someone else), and also encourages you to subclass the frame, which often leads to bad code organization except in the simplest cases.

I think the first steps are to narrow down the methods (or functionality) that we want in the API and to decide where they should go, given the overall organization of our API.

Because less code is better, in some cases we might switch to using methods from external libraries that we already depend on: e.g. `org.apache.commons.lang3.SystemUtils.IS_OS_WINDOWS` instead of `JavaUtils.isWindows()`.

marktsuchida, 2020-08-31:
Classifying as beyond 2.1 for now, but probably appropriate for 2.2 if not 2.1.