micro-manager/micro-manager #760: wrong distances when controlling the stage from keyboard

julou, 2020-04-08:
Hi,

Thanks again to @nicost fro reimplementing the stage control from the keyboard. 
https://github.com/micro-manager/micro-manager/issues/698
https://github.com/micro-manager/micro-manager/pull/741

I could finally go to the lab and test it this morning...
It does work but somehow the movement are downscaled by a lot.

When I shift-arrow (supposed to move by one fov), stage moves by 8.7 instead of 133.79 um. Interestingly, the factor 8.7 / 133.8 = 0.065 which is exactly our pixel size in microns.
Is it possible that the conversion to physical units is done twice? or is is a problem with my config?

Of course I did various  sanity checks  beforehand, in particular that moving the stage from the stage controller plugin or using the mouse-moves-stage work as expected.

Thank you very much for your help. Best,
Thomas

nicost, 2020-04-08:
Thanks for testing (and this is why it is hard to develop without a microscope;).  Problem was pretty obvious and fixed in PR #760.  Hope you can test again!

closed 2020-04-08 by nicost

julou, 2020-04-15:
Thanks @nicost for fixing.

I installed and tested the 20200413 nightly build yesterday… The keyboard control indeed works and now works after the distances set in the stage controller window, great!

I would like to comment on one bug and one usability feature:

- **bug**: in this nightly build, the mouse-moves-stage feature is buggy, with the movement being way too big; I didn't manage to count the precise factor, but I would say at least 10 times too much (this also affects the click-to-center feature). I don't know whether it's related to the stage control form keyboard or whether the bug was simply introduced coincidentally…

- **usability**: in the current version, the 3 text fields used to set distances of stage moves interpret `,` as thousands separator (or maybe ignore them); anyway, I was unfortunate enough that pasting `xx,xxx` from the calculator was interpreted as `xx.xxxe3`. Luckily, 60mm was still not damaging the objective! :-/
I believe that it would be extremely useful to pop up a dialog when the distance entered here are more than 1 or 2 times the camera dimensions, and to print out the formatted distance in microns (and maybe also in mm) in order to give the user a chance to double check. This will not come up too often but might save more than one objective!

nicost, 2020-04-15:
Thanks @julou.  I'll look into both of these.  Copied them into a ticket: #776.  Easier to keep track that way.