micro-manager/micro-manager #766: MMCoreJ should throw a specific exception type instead of `Exception`

nanthony21, 2020-04-10:
The majority of the methods in MMCoreJ are defined with `throws Exception` when really there are probably only a narrow set of exception types that each method can throw. It is generally agreed that it is best to keep try-catch statements as narrowly defined as possible to avoid inadvertently catching unchecked exceptions that you would really like to allow to propagate. An example of this are `InterruptedException`s which are used to cancel an operation running a separate thread. For example image you want to set a property and log an error if it occurs. The obvious thing to write is:

    try {
         doStuff();
         core.setProperty(devName, propName, propVal);
    } catch (Exception e) {
        ReportingUtils.logError(e);
    }

but if this code is run on a separate thread and the thread is interrupted you will find that this broad `catch` statement will catch the `InterruptedException` and log it rather than allow it to actually cancel the thread.

This can potentially be solved by specifically catching any unchecked exceptions that you expect might crop up and then rethrowing them..

    try {
         doStuff();
         core.setProperty(devName, propName, propVal);
    } catch (InterruptedException) {
       Thread.currentThread().interrupt();
       throw new InterruptedException();
    } catch (Exception e) {
        ReportingUtils.logError(e);
    }

However it would be much better if exceptions from MMCoreJ could be narrowly caught allowing unchecked exceptions to automatically propagate:

    try {
         doStuff();
         core.setProperty(devName, propName, propVal);
    } catch (CMMException e) {
        ReportingUtils.logError(e);
    }

nicost, 2020-04-10:
Agreed.  If I remember correctly, it was difficult to configure swig to generate other Exceptions.  Some of the exceptions are hardcoded in MMCoreJ.i, so that should be easy, however, the translation of the C++ CMMError to a specific Java Exception may (or may not) be more difficult.  

I also can not fully oversee if such a change would break other things (at first glance not, since the code is now catching Exception, which should catch everything).  A PR would be great!

marktsuchida, 2020-04-16:
Please see my comments on #772 but I think you can do

```java
try {
     doStuff();
     core.setProperty(devLabel, propName, propVal);
} catch (InterruptedException ie) {
    throw ie;
} catch (Exception e) {
    ReportingUtils.logError(e);
}
```

Admittedly still not ideal.