micro-manager/micro-manager #767: AcquisitionManager "swallows" InterruptedExceptions

nanthony21, 2020-04-10:
In the case that an MDA acquisition is being run as part of a larger routine in a non-GUI thread there may be a case where we want to cancel the routine by calling `Thread.interrupt()` on it's thread. However, if this interrupt is sent in the middle of an MDA acquisition started by `AcquisitionManager.runAcquisition()`, then this line https://github.com/micro-manager/micro-manager/blob/401b9eddb540fdb3133383ff1f96f4f0c5496600/mmstudio/src/main/java/org/micromanager/acquisition/internal/DefaultAcquisitionManager.java#L120 will effectively "swallow" the exception meaning that the code that called `AcquisitionManager.runAcquisition()` will have no idea that the thread has been interrupted and will continue executing.

I would suggest that the `AcquisitionManager` should allow the `InterruptedException` to be thrown so that the outer code can decide how to properly handle this interruption.

Relevant links:
https://www.ibm.com/developerworks/library/j-jtp05236/index.html
https://stackoverflow.com/questions/42219790/which-code-can-swallow-interruptedexception

nicost, 2020-04-10:
That is a change in the API (i.e., corresponding methods in AcquisitionManager will have to throw the InterruptedException).  @marktsuchida  OK to make the change?

marktsuchida, 2020-04-16:
It looks like this code has multiple problems (we shouldn't be polling in a sleep loop), but that does seems like a good change: all blocking methods should have `throws InterruptedException` unless they are theoretically uninterruptible.

I assume the acquisition would need to be stopped when propagating `InterruptedException`.

I would suggest deprecating the variants that take a `boolean shouldBlock` and making separate methods, so that the non-blocking version can be called without checking for `InterruptedException`.