micro-manager/micro-manager #773: Out of Memory calculation in MDA dialog is wrong

nicost, 2020-04-12:
The MDA dialog still assumes that images will be in Direct Memory rather than in the Java memory they are now.

marktsuchida, 2020-04-16:
Also the documentation: https://micro-manager.org/wiki/Micro-Manager_Configuration_Guide#Memory_Settings

Probably nice if the memory warning itself mentions where to set the memory amount.

nicost, 2020-04-17:
Great point!  Was that documentation for 1.4 and is it accurate?  I have no recollection that 1.4 stored all images in Direct Buffers, and a short search shows that utils/ImageUtils/makeTaggedImage() uses Java arrays rather than Direct Buffers.

It would be nice if that is the case, since we then can maintain the same configuration uide for both 1.4 and 2.0 The 2.0 user guide (https://micro-manager.org/wiki/Version_2.0_Users_Guide) points to this general configuration guide for instructions on memory usage.

nicost, 2020-04-17:
Just tested and 1.4 stores images outside of the JVM memory.  It will be difficult to explain which MM version does what, so I am inclined to explain how to figure that out in the documentation.

marktsuchida, 2020-04-17:
`makeTaggedImage()` is not on the path of typical MDA images in 1.4. As I tried to convince you previously, in 1.4 the images are copied to direct byte buffers when storing in RAM, and old 2.0 simply moved that copy upfront. The existing documentation applied to both 1.4 and all versions of 2.0beta/gamma before the recent change.

In 1.4, images coming from the acqEngine are placed (by `DefaultTaggedImageSink`) into an `ImageCache`, whose implementation `MMImageCache` stores images in a `TaggedImageStorage`. The latter has 3 implementations, of which the RAM version is `TaggedImageStorageRamFast`. That class keeps a cache of 10 `TaggedImages` (i.e. heap arrays) but stores the rest in direct byte buffers (in contrast to the non-"Fast" version that is no longer in 1.4). It even goes to the trouble of serializing the metadata into direct byte buffers, and the commit messages from 2013 say that this was also for speed.

This indicates that the limiting factor back then was JVM heap allocation. Maybe the heap needed to be defragmented to allocate large numbers of large arrays, making it slower than off-heap memory, which only requires the operating system to juggle the page tables, but that's just speculation. It should also be noted that the JVM GC is optimized for large numbers of short-lived objects. These things are also likely sensitive to the GC flags passed to the JVM.

It sounds from your experiments that the JVM has improved its heap allocation since (and/or maybe computers just got faster, or maybe it's just because we can now allocate much more memory for the Java heap, and we are ignoring what happens in the 32-bit build) so that this is no longer necessary, but I was taking care to ensure that 2.0 did not change the basic memory usage patterns until we had evidence that it is safe to do so, while trying to move toward a design that would allow us to switch things around more easily by hiding the actual choice of memory behind `Image` (which I'm happy to see pay off now). So the existing documentation should apply to all versions 2013 to March 2020.

I guess a side note is that everything should be coded to work independently of what `Image` uses internally as its buffer. We may in the future have multiple implementations of `Image` that allocate different types of memory (e.g. wrapping direct buffers allocated by the Core or interprocess shared memory). The MDA dialog can of course know which implementation is being used, so can calculate free memory accordingly.

nicost, 2020-04-20:
Fixed the code, needs the documentation fixed too.

nicost, 2020-04-20:
Documentation now fixed too.

closed 2020-04-20 by nicost