micro-manager/micro-manager #802: AcquisitionWrapperEngine saves invalid `SequenceSettings`.

nanthony21, 2020-05-14:
Possibly related to #793

`AcquisitionWrapperEngine` does not load `SequenceSettings` in the same way that is saves them. This appears to be largely due to the fact that `SequenceSettings` do not have boolean fields to mirror the boolean variables of the `AcquisitionWrapperEngine` such as `useFrames_`, `useSlices_`, etc. 

For example, if `useFrames_` is `false` then `AcquisitionWrapperEngine:getSequenceSettings` will set the `numFrames` field of the `SequenceSettings` to 0. However, if those same settings are loaded by `AcquisitionWrapperEngine:setSequenceSettings`, it will set `useFrames_` to `true` and will set `numFrames` to 0, which is an invalid setting which will result in an error when the acquisition is run.

nanthony21, 2020-05-14:
The solution I would propose would be to change `SequenceSettings` so that they fully match the parameters of the `AcquisitionWrapperEngine`. I'm not sure what other effects this might have though.

nicost, 2020-07-18:
This is implemented in PR #882, and seems to work as intended now.  There may be bugs/edge cases, so feel free to re-open if there are still issues.

closed 2020-07-18 by nicost