micro-manager/micro-manager #803: make fails with

strolch101, 2020-05-14:
I am on Linux Mint Debian Edition 4 (Buster / 64bit / gcc 8.3.0 / openJDK 11.0.7 / apache ant 1.10.5), 3rdpartypublic is fetched via svn and available. `./autogen.sh`, `./configure` and `make fetchdeps` ran without errors. However, make fails and keeps giving me the follwing error independent of whether I compile as imagej plugin or not.

`error: package com.bulenkov.iconloader does not exist`

 How do I get rid of it? Thanks.

nicost, 2020-05-14:
When you run make fetchdeps, does the output show any problems?
Is iconloader-GIT.jar present in dependencies/artifacts/compile after running make fetchdeps?
If - for whatever reason - make fetchdpes does not find it, you can copy it manually from 3rdpartypublic/classext
And of course, there may be a completely different problem!

strolch101, 2020-05-14:
I get several "module not found" warnings from `make fetchdeps`:

```
[ivy:resolve]           ::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve]           ::          UNRESOLVED DEPENDENCIES         ::
[ivy:resolve]           ::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-main-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-natives-macosx-universal;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-natives-windows-amd64;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-natives-windows-i586;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-main-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-natives-macosx-universal;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-natives-windows-amd64;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-natives-windows-i586;2.3.2: not found
[ivy:resolve]           :: #iconloader;GIT: not found
[ivy:resolve]           :: #DT1.2;: not found
[ivy:resolve]           ::::::::::::::::::::::::::::::::::::::::::::::
```

Your suggestion to copy iconloader from 3rdpartypublic/classext helped! Now make is complaining about the other missing packages I couldn't locate there I guess. Specifically my build fails now at:

```
[...]
MMClearVolume.compile:
    [mkdir] Created dir: /home/xxx/MicroManager/plugins/build/MMClearVolume
 [mm-javac] Compiling 14 source files to /home/xxx/MicroManager/plugins/build/MMClearVolume
 [mm-javac] warning: [options] bootstrap class path not set in conjunction with -source 8
 [mm-javac] warning: Supported source version 'RELEASE_6' from annotation processor 'com.jogamp.gluegen.structgen.CStructAnnotationProcessor' less than -source '8'
 [mm-javac] warning: No processor claimed any of these annotations: org.scijava.plugin.Plugin,com.google.common.eventbus.Subscribe
 [mm-javac] /home/xxx/MicroManager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/CVViewer.java:24: error: package com.jogamp.newt.awt does not exist
 ...
 ...
 ...
 ...
 [mm-javac] /home/xxx/MicroManager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVSnapshot.java:25: error: cannot find symbol
 [mm-javac]    public boolean screenshot(GLAutoDrawable pGLAutoDrawable,
 [mm-javac]                              ^
 [mm-javac]   symbol:   class GLAutoDrawable
 [mm-javac]   location: class CVSnapshot
 [mm-javac] 7 errors
 [mm-javac] 3 warnings
 
BUILD FAILED
/home/xxx/MicroManager/plugins/build.xml:23: The following error occurred while executing this line:
/home/xxx/MicroManager/plugins/build.xml:7: The following error occurred while executing this line:
/home/xxx/MicroManager/buildscripts/javabuild.xml:112: Compile failed; see the compiler error output for details.

Total time: 20 seconds
make[1]: *** [Makefile:584: plugins.stamp] Fehler 1
make[1]: Verzeichnis „/home/xxx/MicroManager/plugins“ wird verlassen
make: *** [Makefile:521: all-recursive] Fehler 1

```

closed 2020-05-15 by strolch101