micro-manager/micro-manager #817: Using ImageJ Micro-Manager when running MDA causes an error

nicost, 2020-05-27:
Reported by Tronar:

There is an issue, at least with DemoCamera, when using the ROI Manager, and is not related to this plugin but obviously hurts. It happens a lot that starting a second acquisition triggers an error. May be related to multichannel.
Reproduce by :
-open snap
-open ROI manager
-setup a ROI
-run an MDA acquisition
-close / discard
-run another MDA acquisition

java.lang.IndexOutOfBoundsException: Index: 0, Size: 0 in Thread[AWT-EventQueue-0,6,main]
[ ] at java.util.ArrayList.rangeCheck(ArrayList.java:657)
[ ] at java.util.ArrayList.get(ArrayList.java:433)
[ ] at org.micromanager.acquisition.internal.MMAcquisition.(MMAcquisition.java:218)
[ ] at org.micromanager.acquisition.internal.AcquisitionWrapperEngine.runAcquisition(AcquisitionWrapperEngine.java:146)
[ ] at org.micromanager.acquisition.internal.AcquisitionWrapperEngine.acquire(AcquisitionWrapperEngine.java:77)
[ ] at org.micromanager.internal.dialogs.AcqControlDlg.runAcquisition(AcqControlDlg.java:1359)
[ ] at org.micromanager.internal.dialogs.AcqControlDlg.lambda$createRunButtons$22(AcqControlDlg.java:627)

nicost, 2020-06-02:
@tronar: I can not reproduce this (i.e., when I follow the steps above as much as I can, there is no error).  What are the MDA settings?  Is there something you are doing when you "setup a ROI" that I am not doing?

tronar, 2020-06-02:
I've just run it again:
-open MM (demo camera config), snap.
-imageJ, analyze, tools, ROI manager.
-select a rectangle in snap view (Preview), add in Roi Manager.
-open MDA, acquire. (10 time points 1 second apart, 3 channels now (Cy5, DAPI, FITC)
-wait for it, close data window, discard data.
-Aquire again -> Ka boom.
CoreLog attached.
[CoreLog20200602T144001_pid780.txt](https://github.com/micro-manager/micro-manager/files/4718828/CoreLog20200602T144001_pid780.txt)

tronar, 2020-06-06:
Hmm, I guess you are overrun with other issues, but the problem seems to be unrelated to ROI manager. It's kind of hard for me to grasp the whole thing, but the acquisition is assuming that the "current" metadata settings is related to the acquisition, in particular, it has info for all channels.
Break happens in MMAcquisition.java:218 when nsChannels = 3 but "current" displaySettings is related to a single channel.

My MDA settings has channels = 3, but the snap image has only 1 channel. I managed to trigger the issue by doing an acquisition followed by a snap followed by another acquisition.

nicost, 2020-06-06:
I still can not reproduce this.  Can you run this with debug logging enabled, and post the log?

tronar, 2020-06-06:
Sure, mind telling me how to do that ? :)

tronar, 2020-06-06:
Never mind, FAQ did it. But does not happen at once :( I'll try to reproduce it, seems there's a bit of chance in the mix.

tronar, 2020-06-06:
Ok, it did not happen at once, but playing a little did it. 
[CoreLog20200606T192937_pid3648.txt](https://github.com/micro-manager/micro-manager/files/4741088/CoreLog20200606T192937_pid3648.txt)

nicost, 2020-06-07:
Just pushed a likely fix in PR #829.  Since I could never reproduce the problem (either pipeline_.insertSummaryMetadata(summary) could be asynchronous or something else wen wrong, but I think this should fix it.  Not sure if it makes it in tonight's build, but let me know if the problem went away.

tronar, 2020-06-07:
What would be the procedure to know the "current" state of the latest build ? (i.e. if the PR made it)
I've just run into a bit of data that might explain why you were not able to reproduce this. I'm using the 32 bit version, because of compatibility with the Canon "driver".

tronar, 2020-06-07:
Whatever, it seems to work. Thanks. I guess RTIntensities is now on shape to make it to the distribution ?

closed 2020-06-07 by nicost