micro-manager/micro-manager #818: Pixel Size Calibrator silently accepts a configuration that was already defined

nicost, 2020-05-29:
Simplest way to reproduce this using the demo configuration:
- Remove All Pixel Size configurations
- Add a config with only the DObjective label, set it to 10x.
- Create a new config, change the name but leave the DObjective set to 10x. Micro-Manager accepts this setting, but will display 0 nm/px in the main display.  Only one of these two identical configs will be matched.  

It surprises me that the Core allows this setting, and consider that the main bug.

marktsuchida, 2020-09-21:
Fixed, thanks @zhangxiaosa!

nicost, 2020-09-22:
Great that this is fixed in the UI, however, the Core should never allow for this, so this should be fixed there as well.

marktsuchida, 2020-09-22:
Let's make that a separate issue (I'll create it), because the normal config groups have the same problem, not just the pixel size ones.

closed 2020-09-22 by marktsuchida

nicost, 2020-09-22:
Wow, did not realize that.  Thanks!

marktsuchida, 2020-09-22:
@nicost Yeah, I was a little surprised, too, but it is also complicated by how `CMMCore::defineConfig()` works, because each property value is set as a separate function call.