micro-manager/micro-manager #836: bug in core.popNextTaggeImage(int cameraChannelIndex)

henrypinkard, 2020-06-11:
When running a sequence acquisition with multi camera adapter where images are acquired using:

`core_.startSequenceAcquisition(event.getSequence().size(), 0, true);`

and then trying to retrieve images with:

`core_.popNextTaggedImage(camIndex);`

when camIndex is 1, it returns a TaggedImage where all pixels are equal to 0.

A workaround for this is to instead call `core_.popNextTaggedImage()` twice. This returns images from both cameras in a seemingly random order. Which goes to which must be read from the image metadata

nicost, 2020-06-11:
I can not find:
`core_.popNextTaggedImage(camIndex);`
MMCoreJ.i defines:
`public TaggedImage popNextTaggedImage(int cameraChannelIndex) throws java.lang.Exception`
A `cameraChannelIndex `is not the same as `camIndex`. This is all poorly documented, but apparently there are cameras that can deliver multiple channels simultaneously (I think that the multi-photon at Parnassus used this mechanism). The Multicamera device does not use this mechanism, but  tells multiple cameras to start pumping images into the circular buffer.  Your workaround is the "official" way (i.e., the Micro-Manager UI code uses the same mechanism), except that it is not always correct to call popNextImage only twice (it is possible that one camera generates data faster than the other, and you need to call popNextImage more than twice to get data from the "second" camera.  Best to look at the Snap/Live implementation in MM for an approach that seems to work.

closed 2020-06-11 by nicost

henrypinkard, 2020-06-11:
Ah okay. I had assumed multi-camera and multi-channel cameras worked by the same mechanism. I don't think that the number of times calling it will be an issue for me, because I'm simply looping through the number of expected images in the sequence all at once and order doesn't matter.

This raises another question though: how do I figure out how many channels (i.e. cameras) a multi-camera setup has? I assume it is not `core_.getNumberOfCameraChannels()`. Is it `getNumberOfComponents()`? I had previously thought this was only for RGB

nicost, 2020-06-11:
This is all ugly (either the design, or the implementation, or both), and certainly needs better documentation.  I just tried, and getNumberOfCameraChannels() does return 2 for a 2 camera multi-camera configuration, and 1 when I just have a single camera set as the Core Camera.  Yikes!

henrypinkard, 2020-06-11:
Oh well, I think as long as no one has a multi-multi-channel-camera setup it at least works for now