micro-manager/micro-manager #837: build oesnt build

dschneiderch, 2020-06-11:
I can not get micromanager to compile on ubuntu 20.
I am using this [fork](https://github.com/CougPhenomics/micro-manager/tree/ubuntu) where 
plugins/MMClearVolumePlugin/build.xml points directl to the path as specified in #708 and in the [install notes](https://github.com/micro-manager/micro-manager/blob/master/doc/how-to-build.md)

But I'm still getting unresolved dependencies
```
[ivy:resolve]           ::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve]           ::          UNRESOLVED DEPENDENCIES         ::
[ivy:resolve]           ::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-main-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-natives-macosx-universal;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-natives-windows-amd64;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-natives-windows-i586;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-main-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-natives-macosx-universal;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-natives-windows-amd64;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-natives-windows-i586;2.3.2: not found
[ivy:resolve]           :: #DT1.2;: not found
[ivy:resolve]           ::::::::::::::::::::::::::::::::::::::::::::::

```

here is the config. any suggestions @tri/home/raptor/Downloads/make_error.txtccyx  appreciated, thanks.

```
## ---------------------------- ##
## DeviceAdapters configuration ##
## ---------------------------- ##

Note: This is an incomplete list currently under development.

    Build with FreeImagePlus:               yes
    Build with HIDAPI:                      no
    Build with Modbus:                      yes
    Build with libdc1394:                   yes
    Build with libgphoto2:                  yes
    Build with libusb 0.1 (libusb-compat):  yes
    Build with OpenCV:                      no

## --------------------------- ##
## Micro-Manager configuration ##
## --------------------------- ##

   JAVA_HOME    = /usr/lib/jvm/java-11-openjdk-amd64
   JNI_CPPFLAGS = -I/usr/lib/jvm/java-11-openjdk-amd64/include -I/usr/lib/jvm/java-11-openjdk-amd64/include/linux
   JAVA         = /usr/lib/jvm/java-11-openjdk-amd64/bin/java
   JAVACFLAGS   = -Xlint:all,-path,-serial
   ANT          = JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64 /usr/bin/ant

   PYTHON          = /usr/bin/python
   PYTHON_CPPFLAGS =  -I/usr/include/python3.8
   NUMPY_CPPFLAGS  =  -I/home/raptor/.local/lib/python3.8/site-packages/numpy/core/include

   Build MMCoreJ (MMCore Java wrapper):    yes 
   Build MMCorePy (MMCore Python wrapper): yes 
   Build Java application (MMStudio):      yes 
   Build device adapters:                  yes
   Build closed-source device adapters:    no

   Install MMStudio as ImageJ plugin:      yes
   Target ImageJ directory:                /home/raptor/Documents/GitHub/mm_scripts/Fiji.app
   ij.jar:                                 /home/raptor/Documents/GitHub/mm_scripts/Fiji.app/jars/ij-1.52p.jar
   Install dependency JARs:                yes

First type 'make fetchdeps' once to download Java build dependencies.
Type 'make' to build.
Then type 'make install' to install.

```
Full compile log:
[make_error.txt](https://github.com/micro-manager/micro-manager/files/4767256/make_error.txt)

nicost, 2020-06-11:
It looks like you did not pull 3rdpartypublic and put it next to your micro-manager source:
`svn checkout https://valelab4.ucsf.edu/svn/3rdpartypublic/`
You can also just ignore these errors and type make.  The ClearVolume plugin should build since you have the dependencies setup correctly, DT is for the Magellan plugin. 

B.t.w., if anyone sends a PR for the linux build that I can apply without breaking the Mac and WIndows build, than I'll happily apply it;)

dschneiderch, 2020-06-11:
@nicost  I did pull the 3rdpartypublic

```
raptor@cppc2:~/Documents/GitHub/mm_scripts$ ls
3rdpartypublic  Fiji.app          LICENSE        README.md  std.txt
build-mm.sh     fiji-linux64.zip  micro-manager  scripts
raptor@cppc2:~/Documents/GitHub/mm_scripts$ find 3rdpartypublic
3rdpartypublic
3rdpartypublic/boost-linux-x86_64
3rdpartypublic/classext
3rdpartypublic/classext/clooj.jar
3rdpartypublic/classext/iconloader.jar
3rdpartypublic/classext/TSFProto.jar
```

build script is here: https://github.com/CougPhenomics/mm_scripts/blob/master/build-mm.sh

dschneiderch, 2020-06-11:
If I do 
```
make clean
make
```
I still get an error on MMCoreJ_wrap

```
MMCoreJ.compile:
    [mkdir] Created dir: /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/build
 [mm-javac] Compiling 45 source files to /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/build
 [mm-javac] warning: [options] bootstrap class path not set in conjunction with -source 8
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONTokener.java:373: warning: [deprecation] Integer(int) in Integer has been deprecated
 [mm-javac]                         return new Integer(Integer.parseInt(s.substring(2),
 [mm-javac]                                ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONTokener.java:380: warning: [deprecation] Integer(int) in Integer has been deprecated
 [mm-javac]                         return new Integer(Integer.parseInt(s, 8));
 [mm-javac]                                ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONTokener.java:387: warning: [deprecation] Integer(String) in Integer has been deprecated
 [mm-javac]                 return new Integer(s);
 [mm-javac]                        ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONTokener.java:390: warning: [deprecation] Long(String) in Long has been deprecated
 [mm-javac]                     return new Long(s);
 [mm-javac]                            ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONTokener.java:393: warning: [deprecation] Double(String) in Double has been deprecated
 [mm-javac]                         return new Double(s);
 [mm-javac]                                ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONArray.java:543: warning: [deprecation] Double(double) in Double has been deprecated
 [mm-javac]         Double d = new Double(value);
 [mm-javac]                    ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONArray.java:557: warning: [deprecation] Integer(int) in Integer has been deprecated
 [mm-javac]         put(new Integer(value));
 [mm-javac]             ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONArray.java:569: warning: [deprecation] Long(long) in Long has been deprecated
 [mm-javac]         put(new Long(value));
 [mm-javac]             ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONArray.java:613: warning: [deprecation] Double(double) in Double has been deprecated
 [mm-javac]         put(index, new Double(value));
 [mm-javac]                    ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONArray.java:628: warning: [deprecation] Integer(int) in Integer has been deprecated
 [mm-javac]         put(index, new Integer(value));
 [mm-javac]                    ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONArray.java:643: warning: [deprecation] Long(long) in Long has been deprecated
 [mm-javac]         put(index, new Long(value));
 [mm-javac]                    ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONObject.java:93: warning: [overrides] Class Null overrides equals, but neither it nor any superclass overrides hashCode method
 [mm-javac]      private static final class Null {
 [mm-javac]                           ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONObject.java:569: warning: [deprecation] Double(String) in Double has been deprecated
 [mm-javac]                 new Double((String)o).doubleValue();
 [mm-javac]                 ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONObject.java:720: warning: [deprecation] Double(double) in Double has been deprecated
 [mm-javac]         put(key, new Double(value));
 [mm-javac]                  ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONObject.java:734: warning: [deprecation] Integer(int) in Integer has been deprecated
 [mm-javac]         put(key, new Integer(value));
 [mm-javac]                  ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONObject.java:748: warning: [deprecation] Long(long) in Long has been deprecated
 [mm-javac]         put(key, new Long(value));
 [mm-javac]                  ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/src/main/java/mmcorej/org/json/JSONWriter.java:293: warning: [deprecation] Double(double) in Double has been deprecated
 [mm-javac]         return this.value(new Double(d));
 [mm-javac]                           ^
[... cut ...]
 [mm-javac] Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
 [mm-javac] 8 errors
 [mm-javac] 31 warnings

BUILD FAILED
/home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/build.xml:15: Compile failed; see the compiler error output for details.

Total time: 2 seconds
make[1]: *** [Makefile:816: MMCoreJ.jar] Error 1
make[1]: Leaving directory '/home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap'
make: *** [Makefile:521: all-recursive] Error 1
raptor@cppc2:~/Documents/GitHub/mm_scripts/micro-manager$ 
```

dschneiderch, 2020-06-11:
this seems relevant to the os problem:
https://alvinalexander.com/blog/post/java/how-determine-operating-system-os-ant-build-script/

dschneiderch, 2020-06-12:
@nicost It looks like there is a whole series of these type casting errors in MMCoreJ_wrap that might be tripping me up

```
[mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:231: error: incompatible types: Long cannot be converted to int
 [mm-javac]          Rectangle r = new Rectangle((int) xs.get(i), (int) ys.get(i),
```

dschneiderch, 2020-06-12:
I don't know anything about java or c++ but  I'd guess this is the same issue:
https://stackoverflow.com/questions/10596052/stdvector-to-java-util-vector-code-generation-with-swig?noredirect=1&lq=1
https://stackoverflow.com/questions/8145605/swig-v1-3-29-generated-c-to-java-vector-class-not-acting-properly/8190135#8190135

https://github.com/micro-manager/micro-manager/blob/b0bde3d6b1fa91be2ea0ec44e8ad9ec03fe52b22/MMCoreJ_wrap/MMCoreJ.i#L345

I don't know if thats helpful but at this point i'm way over my head.

nicost, 2020-06-12:
Yes, that seems your current key problem.  Do you know which Swig version you are running?

Can you look in the file MMCoreJ_wrap/gensrc/mmcorej/UnsignedVector.java and look for the function: 
`public long get(int i)`
I suspect that your Swig version generates:
`public Long get(int i)`
and if that is all true, you could check this by changing that one letter manually.  
Not sure how to solve this all elegantly, but it sure is time for MM to update the build  toolsets (C++/Java compilers, and SWIG), which should help building the code on newer Linux distributions.

dschneiderch, 2020-06-12:
I"m using swig 4.0.1 and you are correct. however changing Long to long ends up with conflicts with other definitions (i think)
```
Buildfile: /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/build.xml

MMCoreJ.compile:
 [mm-javac] Compiling 18 source files to /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/build
 [mm-javac] warning: [options] bootstrap class path not set in conjunction with -source 8
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/UnsignedVector.java:16: error: UnsignedVector is not abstract and does not override abstract method get(int) in AbstractList
 [mm-javac] public class UnsignedVector extends java.util.AbstractList<Long> implements java.util.RandomAccess {
 [mm-javac]        ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/UnsignedVector.java:95: error: get(int) in UnsignedVector cannot override get(int) in AbstractList
 [mm-javac]   public long get(int index) {
 [mm-javac]               ^
 [mm-javac]   return type long is not compatible with Long
 [mm-javac]   where E is a type-variable:
 [mm-javac]     E extends Object declared in class AbstractList
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/UnsignedVector.java:99: error: set(int,Long) in UnsignedVector cannot implement set(int,E) in List
 [mm-javac]   public long set(int index, Long e) {
 [mm-javac]               ^
 [mm-javac]   return type long is not compatible with Long
 [mm-javac]   where E is a type-variable:
 [mm-javac]     E extends Object declared in interface List
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/UnsignedVector.java:72: warning: [cast] redundant cast to int
 [mm-javac]          Long ints[] = new Long[(int) size()];
 [mm-javac]                                 ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/gensrc/mmcorej/StrMap.java:66: warning: [cast] redundant cast to String
 [mm-javac]     Iterator itr = find((String) key);
 [mm-javac]                         ^
 [mm-javac] 3 errors
 [mm-javac] 3 warnings

BUILD FAILED
/home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap/build.xml:15: Compile failed; see the compiler error output for details.

Total time: 1 second
make[1]: *** [Makefile:816: MMCoreJ.jar] Error 1
make[1]: Leaving directory '/home/raptor/Documents/GitHub/mm_scripts/micro-manager/MMCoreJ_wrap'
make: *** [Makefile:521: all-recursive] Error 1

```

 Looks like ubuntu 18 LTS is on 3.0.12 so i will try to downgrade swig since it worked before.

nicost, 2020-06-12:
Yes, it looks like Swig made some improvements (the new code is much nicer and easier to work with on the Java side), but the change makes code written for the code produced by the old Swig version incompatible with the new one.  If the downgrading does not work or is hard, I have an idea how to fix this in MMcoreJ.i (a bit too busy right now, but I can post the code idea here later).

dschneiderch, 2020-06-12:
swig 3.0 did the trick. Thanks for your help! Wish I could help with the underlying cause.
To get swig 3.0.12 I was able to use:
```
sudo apt install swig3.0
sudo update-alternatives --install /usr/bin/swig swig /usr/bin/swig3.0 1
sudo update-alternatives --install /usr/bin/swig swig /usr/bin/swig4.0 2
sudo update-alternatives --config swig
```
then choose the version you want to use

closed 2020-06-12 by dschneiderch

nicost, 2020-06-16:
I made a modified build.xml (for plugins/MMClearVolume) that has the linux-specific path in a conditional manner.  Can you check that it works as expected on linux (https://github.com/nicost/micro-manager/blob/linuxMMCV/plugins/MMClearVolumePlugin/build.xml).  If it does, I'll merge it in, so that you and others will have an easier time next time you compile MM.

dschneiderch, 2020-06-16:
Nope.

`make fetchdeps` still gives
```
[ivy:resolve]           ::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve]           ::          UNRESOLVED DEPENDENCIES         ::
[ivy:resolve]           ::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-main-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-natives-macosx-universal;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-natives-windows-amd64;2.3.2: not found
[ivy:resolve]           :: org.jogamp.gluegen#gluegen-rt-natives-windows-i586;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-main-local;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-natives-macosx-universal;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-natives-windows-amd64;2.3.2: not found
[ivy:resolve]           :: org.jogamp.jogl#jogl-all-natives-windows-i586;2.3.2: not found
[ivy:resolve]           :: #DT1.2;: not found
[ivy:resolve]           ::::::::::::::::::::::::::::::::::::::::::::::
[ivy:resolve] 
[ivy:resolve] :: USE VERBOSE OR DEBUG MESSAGE LEVEL FOR MORE DETAILS
[ivy:retrieve] :: retrieving :: org.micromanager#micromanager [sync]
[ivy:retrieve]  confs: [build, test, compile, optional, imagej, runtime]
[ivy:retrieve]  conflict on /home/raptor/Documents/GitHub/mm_scripts/micro-manager/dependencies/artifacts/compile/coremem-0.4.5.jar in [compile]: 0.4.5 won
[ivy:retrieve]  0 artifacts copied, 135 already retrieved (0kB/40ms)
    [mkdir] Created dir: /home/raptor/Documents/GitHub/mm_scripts/micro-manager/dependencies/artifacts/optional

BUILD SUCCESSFUL
Total time: 46 seconds
```
and `make` ends with
```
MMClearVolume.compile:
    [mkdir] Created dir: /home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/build/MMClearVolume
 [mm-javac] Compiling 14 source files to /home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/build/MMClearVolume
 [mm-javac] warning: [options] bootstrap class path not set in conjunction with -source 8
 [mm-javac] warning: Supported source version 'RELEASE_6' from annotation processor 'com.jogamp.gluegen.structgen.CStructAnnotationProcessor' less than -source '8'
 [mm-javac] warning: No processor claimed any of these annotations: org.scijava.plugin.Plugin,com.google.common.eventbus.Subscribe
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/CVViewer.java:24: error: package com.jogamp.newt.awt does not exist
 [mm-javac] import com.jogamp.newt.awt.NewtCanvasAWT;
 [mm-javac]                           ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/CVViewer.java:28: error: package com.jogamp.newt does not exist
 [mm-javac] import com.jogamp.newt.NewtFactory;
 [mm-javac]                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVVideoRecorder.java:5: error: package com.jogamp.opengl does not exist
 [mm-javac] import com.jogamp.opengl.GL;
 [mm-javac]                         ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVVideoRecorder.java:6: error: package com.jogamp.opengl does not exist
 [mm-javac] import com.jogamp.opengl.GLAutoDrawable;
 [mm-javac]                         ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVVideoRecorder.java:23: error: cannot find symbol
 [mm-javac]    public boolean screenshot(GLAutoDrawable pGLAutoDrawable,
 [mm-javac]                              ^
 [mm-javac]   symbol:   class GLAutoDrawable
 [mm-javac]   location: class CVVideoRecorder
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVSnapshot.java:5: error: package com.jogamp.opengl does not exist
 [mm-javac] import com.jogamp.opengl.GL;
 [mm-javac]                         ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/MMClearVolumePlugin/src/main/java/edu/ucsf/valelab/mmclearvolumeplugin/recorder/CVSnapshot.java:25: error: cannot find symbol
 [mm-javac]    public boolean screenshot(GLAutoDrawable pGLAutoDrawable,
 [mm-javac]                              ^
 [mm-javac]   symbol:   class GLAutoDrawable
 [mm-javac]   location: class CVSnapshot
 [mm-javac] 7 errors
 [mm-javac] 3 warnings

BUILD FAILED
/home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/build.xml:23: The following error occurred while executing this line:
/home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins/build.xml:7: The following error occurred while executing this line:
/home/raptor/Documents/GitHub/mm_scripts/micro-manager/buildscripts/javabuild.xml:112: Compile failed; see the compiler error output for details.

Total time: 12 seconds
make[1]: *** [Makefile:584: plugins.stamp] Error 1
make[1]: Leaving directory '/home/raptor/Documents/GitHub/mm_scripts/micro-manager/plugins'
make: *** [Makefile:521: install-recursive] Error 1
raptor@cppc2:~/Documents/GitHub/mm_scripts/micro-manager$ 
```