micro-manager/micro-manager #838: micromanager won't compile with swig 4.x

dschneiderch, 2020-06-12:
see #837 

to see your swig version:
`swig -version`

current work around is to install swig 3.0.12

On Ubuntu 20.04 LTS (default swig is 4.x) I was able to use:
```
sudo apt install swig3.0
sudo update-alternatives --install /usr/bin/swig swig /usr/bin/swig3.0 1
sudo update-alternatives --install /usr/bin/swig swig /usr/bin/swig4.0 2
sudo update-alternatives --config swig
```
then choose the version you want to use and attempt to compile micromanager.
Note, on ubuntu 18+ you need to hardcode jogl2.jar path in plugins/MMClearVolume/build.xml by replacing current contents with the following (see #708 )
```
<project name="MMClearVolume" basedir="." default="jar">
   <import file="../javapluginbuild.xml" />
   <path id="plugin.classpath">
           <pathelement location="/usr/share/java/jogl2.jar" />
   </path>
</project>
```

nicost, 2020-06-15:
I made some changes to MMCoreJ.i that still compile with swig 3, but I believe to also compile with Swig 4.  You can find the branch at github.com/nicost/micro-manager under the name swig4.  Or you can copy the file MMCoreJ.i (https://github.com/nicost/micro-manager/blob/swig4/MMCoreJ_wrap/MMCoreJ.i) into your current source code.  You may need to do a make clean in the upper build directory to ensure that swig will run again.  Curious to hear if that compiles OK with swig 4 generated code (if so, I'll push this into the main branch). I will also look at the ant issue in the CLearVolume plugin.

dschneiderch, 2020-06-15:
I copied the new file into my directory, switched back to swig4.0, ran make clean, then make clean, ./autogen.sh, make and get similar error.

```
Making all in mmstudio
make[1]: Entering directory '/home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio'
JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64 /usr/bin/ant -Dmm.javacflags="-Xlint:all,-path,-serial" -logger org.apache.tools.ant.listener.SimpleBigProjectLogger jar
Buildfile: /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/build.xml

mmstudio.compile:
    [mkdir] Created dir: /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/build
 [mm-javac] Compiling 574 source files to /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/build
 [mm-javac] warning: [options] bootstrap class path not set in conjunction with -source 8
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/acquisition/internal/MMAcquisition.java:64: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/OMEMetadata.java:51: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/MultipageTiffReader.java:59: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/display/internal/DefaultDisplaySettings.java:42: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/internal/utils/imageanalysis/ImageUtils.java:19: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] warning: Supported source version 'RELEASE_6' from annotation processor 'com.jogamp.gluegen.structgen.CStructAnnotationProcessor' less than -source '8'
 [mm-javac] warning: No processor claimed any of these annotations: org.scijava.plugin.Plugin,org.micromanager.internal.utils.MustCallOnEDT,com.google.common.eventbus.Subscribe
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/acquisition/internal/MMAcquisition.java:64: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/OMEMetadata.java:51: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/MultipageTiffReader.java:59: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/display/internal/DefaultDisplaySettings.java:42: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/internal/utils/imageanalysis/ImageUtils.java:19: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/acquisition/internal/MMAcquisition.java:64: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
 [mm-javac] /home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/OMEMetadata.java:51: warning: [deprecation] MDUtils in org.micromanager.internal.utils has been deprecated
 [mm-javac] import org.micromanager.internal.utils.MDUtils;
 [mm-javac]                                       ^
[... cut ...]
BUILD FAILED
/home/raptor/Documents/GitHub/mm_scripts/micro-manager/buildscripts/javabuild.xml:112: Compile failed; see the compiler error output for details.

Total time: 10 seconds
make[1]: *** [Makefile:892: MMJ_.jar] Error 1
make[1]: Leaving directory '/home/raptor/Documents/GitHub/mm_scripts/micro-manager/mmstudio'
make: *** [Makefile:521: all-recursive] Error 1
raptor@cppc2:~/Documents/GitHub/mm_scripts/micro-manager$ 
                                                                          
```

nicost, 2020-06-15:
Perfect!  It got much further and hit other code that has similar issues.  I'll look into fixing those as well (pretty hard without having Swig 4 available, and no desire to try to install it on my Windows box ;)

nicost, 2020-06-15:
It would help to see the file DoubleVector.java that is generated by your version of Swig.  Any way you can get that to me?

dschneiderch, 2020-06-15:
No Problem. here is the entire mmcorej_wrap folder that swig creates.
[MMCoreJ_wrap.zip](https://github.com/micro-manager/micro-manager/files/4783369/MMCoreJ_wrap.zip)

nicost, 2020-06-15:
It looks like the Swig folks very reworked the code translating C++ STL data structures into Java code, and I see no way to write Java code that is compatible with both Swig 3 and 4 output.  Pity, since the MM code barely uses it.  On the other hand, it is easy for the time being to use Swig < 4 (as you do), and it also will be easy to make the changes to use Swig 4, once all supported platforms are ready to move to Swig >= 4.0.  So, I'll close this issue, since it turns out to be impossible to support Swig < 4 and Swig >= 4 at the same time.

closed 2020-06-15 by nicost