micro-manager/micro-manager #857: Events in Micro-Manager are difficult to understand and work with

nicost, 2020-06-27:
There are a couple of different event busses, one emanating from Studio, one from the DisplayManager, and one from the DataManager.  Event types are scattered through the code, and it is not always clear where to register for a specific event.  On top of that, quite a few useful Events are internal, so should not be used by plugins. 

- Would it be possible to have just one Event bus rather than three? 
- Can we keep a list of Events (ordered by the Event bus that passes them on if we stick to 3 busses) somewhere?   That could well be in the code itself.
- Can all internal Events be evaluated, and can we make as many as possible of those external?

nanthony21, 2020-07-07:
On a related note, the "ConfigGroupChangedEvent" doesn't appear to behave as expected. For example, in the "MMConfig_demo.cfg" configuration:
 - The "Channel" and "LightPath" configurations fire no events.
 - Changing the "Objective" configuration group fires a "System" event but no "Objective" event.
 - Changing the "System" configuration groups fires multiple "System" and "Camera" events even when the change also involves a change to the "Objective" group.

marktsuchida, 2020-08-31:
There is no event bus associated with `DisplayManager` or `DataManager`. There are separate buses for each display and datastore, to which user code can subscribe as needed. Because subscriptions are done based on the event class, using a single event bus would mean that any object subscribed to be notified (for example) of new images in a particular datastore would get called any time an image is added to any datastore. This would not be acceptable when a fast acquisition is running in the presence of other (finished) datasets being open. It's also easier to write code that deals with a single display or datastore (which is the most common case) if you don't have to check each event for a match.

The `ConfigGroupChangedEvent` issue is unrelated -- it has to do with problems with the design of the MMCore and MMDevice property API and I don't think it can be fundamentally fixed, although we might be able to work around it better.

The design, naming, and documentation for individual events is in fact messy and needs work.
- Some need to be renamed to make sense.
- Many need documentation to precisely specify semantics.
- Some need to be moved to a sensible package.
- API events, like other API types, should be interfaces, not classes (consider whether it is okay to keep the exception made for core callback events).
- Some events that should be in the API are missing (please file individual tickets if you encounter any, referencing this issue if it's still open).
- We should provide tag interfaces that mirror the event bus on which the event is posted. For example, all events posted (by the API) to a display's event bus should derive from `DisplayEvent`. This will allow us to browse the Javadoc (which lists "all known subinterfaces") to discover all available event types. It also provides a consistent way to access the display or datastore to which the event applies.

(Note that I simply wrote datastore above where actually it might be data provider, etc.)

marktsuchida, 2020-11-18:
Superseded by #1018.

closed 2020-11-18 by marktsuchida