micro-manager/micro-manager #858: NikonTI PFS does not respond to CMMCore::get/setAutofocusOffset

nanthony21, 2020-06-27:
Using the setter for `AutofocusOffset`  has no effect and the getter always returns 0 even though `CMMCore::enableContinuousFocus` works fine.

Even though there is a `property` that can be used to adjust the offset it would be nice if the API was implemented so that autofocusing code doesn't need to take special consideration for PFS to work.

As a sidenote, is there any sort of way to get some sort of error signal from the PFS or other hardware focus devices? I'm trying to find a way to know how close the PFS is to being locked even when continuous focus is not enabled .

nicost, 2020-07-03:
I do remember that the PFS offset was hard to work with since the number it reports is only a relative, dimensionless number.  So, what does it mean when someone calls the SetAutofocusOffset function?  Setting it to 0 and using that to set the PFS Offset would be disastrous.  
The real problem is that hardware autofocus devices present themselves differently.  I have worked a bit on the code for the Zeiss Definite focus and the Nikon PFS, and they have quite different interfaces, and the Olypmpus and Leica ones manage to do it yet another way.  So, abstraction is hard.  However, the Core interface was designed very early on, and it makes total sense to start thinking what a better interface should look like (by looking carefully at the implementations of the big 4, the ASI CRISP, and possibly others?)

nanthony21, 2020-07-05:
PFS on the TI and TI2 are all that I've worked with so I can't comment on other hardware focus lock devices but I think that the current API suits the Nikon devices quite well, it's just that in the case of the TI this API was never actually implemented.

In my own plugin I have some code that performs a calibration to convert the "offset" to microns so that the Nikon Z stage position can be set in units of microns regardless of whether PFS is enabled. Maybe this would be a useful thing to share with others though I'm not sure what would be the proper way to package it for general usage. Do you know if there is any existing way to register a Java class so that it is treated just like a device adapter?

marktsuchida, 2020-08-31:
As Nico wrote, achieving a vendor-neutral interface will take much more than having a getter/setter for offset; in fact even for the devices that have a numerical offset, they do not mean the same thing. And modeling the offset of a hardware focus-maintenance device as a pseudo-z-drive was probably a bad idea.

That said, if by "setter for `AutofocusOffset`" you are talking about `CMMCore::setPosition` called on the `PFS Offset` device, it should work (possibly only when continuous focus is enabled). Can you provide the exact script/code that failed?

As for error signals, the several possible values of the `Status` property are all that is available.

nanthony21, 2020-09-01:
`CMMCore::setPosition` and `getPosition` do in fact work for the TI. 

I suppose my confusion just came from the fact that the `CMMCore::get/setAutofocusOffset` methods accurately describe how adjusting the PFS works yet it wasn't implemented.

It sounds like the consensus is that the existing `autofocusOffset` methods are not general purpose enough and should probably  be deprecated.

closed 2020-09-01 by nanthony21

marktsuchida, 2020-09-01:
Oh, I see. Agreed!