micro-manager/micro-manager #862: Autofocus fails during `live mode`

nanthony21, 2020-06-30:
I have seen this issue with OughtaFocus and JAF(H&P) though it may also occur with other autofocus plugins.

When live mode is enabled if you run an autofocus it will always select the original z position as the optimal position.

Here is the log from running OughtaFocus with live mode off:

```
2020-06-30T16:01:42.295567 tid1192 [IFO,App] OughtaFocus: image=0, t=1181, z=3624.75, score=0.10, Tz=928, Ti=201, Tc=28
2020-06-30T16:01:43.162616 tid1192 [IFO,App] OughtaFocus: image=1, t=2048, z=3622.84, score=0.08, Tz=637, Ti=200, Tc=30
2020-06-30T16:01:44.395687 tid1192 [IFO,App] OughtaFocus: image=2, t=3281, z=3626.66, score=0.14, Tz=1004, Ti=201, Tc=28
2020-06-30T16:01:44.970720 tid1192 [IFO,App] OughtaFocus: image=3, t=3856, z=3627.84, score=0.19, Tz=346, Ti=201, Tc=28
2020-06-30T16:01:45.355742 tid1192 [IFO,App] OughtaFocus: image=4, t=4241, z=3628.57, score=0.24, Tz=156, Ti=201, Tc=28
2020-06-30T16:01:46.045781 tid1192 [IFO,App] OughtaFocus: image=5, t=4931, z=3629.02, score=0.29, Tz=461, Ti=200, Tc=29
2020-06-30T16:01:46.601813 tid1192 [IFO,App] OughtaFocus: image=6, t=5487, z=3629.30, score=0.32, Tz=327, Ti=200, Tc=29
2020-06-30T16:01:47.096841 tid1192 [IFO,App] OughtaFocus: image=7, t=5982, z=3629.47, score=0.34, Tz=266, Ti=200, Tc=29
2020-06-30T16:01:47.530866 tid1192 [IFO,App] OughtaFocus: image=8, t=6416, z=3629.58, score=0.35, Tz=204, Ti=201, Tc=29
2020-06-30T16:01:47.530866 tid1192 [IFO,App] OughtaFocus Iterations: 8, z=3629.58, dz=4.83, t=6416
```

and here is a similar log when live mode is on:

```
2020-06-30T16:01:06.979547 tid10016 [IFO,App] OughtaFocus: image=0, t=103, z=3619.95, score=0.07, Tz=49, Ti=2, Tc=28
2020-06-30T16:01:07.165558 tid10016 [IFO,App] OughtaFocus: image=1, t=289, z=3618.04, score=0.07, Tz=156, Ti=2, Tc=28
2020-06-30T16:01:07.370569 tid10016 [IFO,App] OughtaFocus: image=2, t=494, z=3616.86, score=0.07, Tz=175, Ti=2, Tc=28
2020-06-30T16:01:07.889599 tid10016 [IFO,App] OughtaFocus: image=3, t=1013, z=3616.13, score=0.07, Tz=490, Ti=1, Tc=28
2020-06-30T16:01:08.475632 tid10016 [IFO,App] OughtaFocus: image=4, t=1599, z=3615.68, score=0.07, Tz=557, Ti=2, Tc=27
2020-06-30T16:01:08.749648 tid10016 [IFO,App] OughtaFocus: image=5, t=1873, z=3615.40, score=0.07, Tz=244, Ti=2, Tc=28
2020-06-30T16:01:08.905657 tid10016 [IFO,App] OughtaFocus: image=6, t=2029, z=3615.23, score=0.07, Tz=127, Ti=1, Tc=28
2020-06-30T16:01:09.630699 tid10016 [IFO,App] OughtaFocus: image=7, t=2754, z=3615.12, score=0.07, Tz=695, Ti=2, Tc=28
2020-06-30T16:01:09.630699 tid10016 [IFO,App] OughtaFocus Iterations: 7, z=3619.95, dz=0.00, t=2754
```

You can see that even though the Z position is changing the `score` always remains constant. This makes me think that maybe when live mode is enabled the autofocus plugins end up just analyzing the same image over and over again rather than snapping a new image for each position.

On a related note, I was wondering how to select the best autofocus plugin for my application. Is there one that is generally more reliable than the others?

nicost, 2020-07-03:
I can not be of much help here as I never use these Java software autofocus routines. They are also a bit difficult to trouble shoot without actual hardware. The JAF(H&P) was one of the very first.  Arthur Edelstein wrote the Oughtafocus, which suggests that it ought to focus;)  If I remember correctly, the Oughtafocus plugin has quite a few settings, so whether or not it works in real life will depend on how you set it up.

nanthony21, 2020-07-05:
I'm guessing that this is just a matter of making sure to suspend live mode before the autofocus begins. I'll try to take a look at it this week.

nanthony21, 2020-07-06:
So it looks like this comes down to a mismatch between how "OughtaFocus" expects live mode to be handled and how `MMStudio::autofocusNow` actually handles it. "OughtaFocus" is written to work even while live mode is running so it gets its images using.

```
         if (liveModeOn_) {
            img = core.getLastTaggedImage();
         } else {
            core.waitForDevice(core.getCameraDevice());
            core.snapImage();
```

However, MMStudio makes sure to suspend livemode before starting the autofocus. Apparently with livemode in a suspended state `Studio.live().getIsLiveModeOn` still evaluates as true. "OughtaFocus" ends up pulling the same image from the queue over and over again which prevents any meaningful focusing from happening.

I'm guessing we don't want to change `MMStudio::autofocusNow` so it's probably best to fix this in "OughtaFocus" and any other autofocus that has this issue. I also wonder if it is correct for `Studio.live().getIsLiveModeOn`  to be considered `true` when live mode is suspended.

nicost, 2020-07-06:
I don't understand the intent of the Oughtafocus code with respect to using live mode images.  That is certainly asking for trouble!

I agree that the Live Mode api is ambiguous. The comment to getIsLiveModeOn is: "Indicates if live mode is currently running.".  The idea is also to make it easy to suspend live mode.  Maybe adding an `isLiveModeSuspended` function is the solution?  @marktsuchida  what do you think?

marktsuchida, 2020-09-01:
I don't think we can easily get autofocus to work during Live without suspending it, because there is no way to synchronize Z movement with the images. So I don't think we have any option other than for Oughtafocus to always call `snapImage()`, under the (correct) assumption that Live will be suspended by MMStudio.

The only better way I can think of is to use software triggers, but that requires changing how everything in MM works.