micro-manager/micro-manager #868: Consider changing blacklist to blocklist

ctrueden, 2020-07-03:
See https://twitter.com/leahculver/status/1269109776983547904

nicost, 2020-07-03:
Thanks for the poke @ctrueden ! I found one in the SerialManager device adapater and one in an obscure buildscript.  Was there any specific instance you are aware of, or was this more in general?  I only searched the source, if you encountered this in any documentation, please let me know.

ctrueden, 2020-07-03:
@nicost I found those instances in the source as well, but didn't check the docs. I did a mass grep over many repositories I track, and filed issues in places where I noticed them.

nicost, 2020-07-03:
Thanks for doing that Curtis!  Closing now as this is fixed in svn, will sing through to the github sources soon.

closed 2020-07-03 by nicost