micro-manager/micro-manager #890: Intermitent failure to start acquisition

impact27, 2020-07-28:
I am using pycromanager to control micromanager and start acquisition. Using gama 20200527, I sometimes have the following problem when starting an acquisition:

I have this code to start an acquisition:
``` python
from pycromanager import Bridge
import os.path

bridge = Bridge(convert_camel_case=False)
mmStudio = bridge.get_studio()
acquisition_manager = mmStudio.acquisitions()
snapLiveManager = mmStudio.live()
displayManager = mmStudio.displays()

def start_acquisition(
        save_dir, experiment_name,
        numFrames, interval=0, comments=''):
    """Start acquisition using acquisition engine."""
    if acquisition_manager.isAcquisitionRunning():
        raise RuntimeError("Acquisition already running")

    save_dir = os.path.abspath(os.path.expanduser(save_dir))
    os.makedirs(save_dir, exist_ok=True)

    settings = bridge.construct_java_object(
        'org.micromanager.acquisition.SequenceSettings')
    settings.intervalMs = int(interval  * 1000)
    settings.numFrames = numFrames
    settings.prefix = experiment_name
    settings.root = save_dir
    settings.save = True
    settings.shouldDisplayImages = True
    settings.comment = comments

    datastore = acquisition_manager.runAcquisitionWithSettings(settings, False)
    return datastore
```

Sometimes, the acquisition fails to start, an error windows appears on micro manager infoming me that : 
`Index: 0, Size: 0`

The return value of `acquisition_manager.runAcquisitionWithSettings` is None and `acquisition_manager.isAcquisitionRunning()` is True.

So what I think is happening is that an exception is raised somewhere in the micromanager code. Because I don't get an exception but instead a `None`, I suspect that the curlpit is:

https://github.com/micro-manager/micro-manager/blob/d455aacce58721e8442d99bca9630106732c17b6/mmstudio/src/main/java/org/micromanager/acquisition/internal/AcquisitionWrapperEngine.java#L198

I can not easily reproduce because the error is quite rare.

marktsuchida, 2020-07-28:
The `Index: 0, Size: 0` error message suggests that it is an `IndexOutOfBoundsException`. Possibly related is #817, which suggests one possible source of such an error, although there could easily be others.

In most cases, the stack trace corresponding to the error is logged in the CoreLog. If you can find that the next time you encounter this error (preferably with debug logging enabled in Tools `>` Options), it will be the most useful information to track down the cause.

impact27, 2020-07-29:
Got it. The traceback is:
```
2020-07-29T09:48:13.305761 tid1664 [IFO,App] 
                                   [       ] java.lang.IndexOutOfBoundsException: Index: 0, Size: 0 in Thread[ZMQ Server ,6,main]
                                   [       ]   at java.util.ArrayList.rangeCheck(ArrayList.java:657)
                                   [       ]   at java.util.ArrayList.get(ArrayList.java:433)
                                   [       ]   at org.micromanager.acquisition.internal.MMAcquisition.<init>(MMAcquisition.java:218)
                                   [       ]   at org.micromanager.acquisition.internal.AcquisitionWrapperEngine.runAcquisition(AcquisitionWrapperEngine.java:146)
                                   [       ]   at org.micromanager.acquisition.internal.AcquisitionWrapperEngine.acquire(AcquisitionWrapperEngine.java:77)
                                   [       ]   at org.micromanager.acquisition.internal.DefaultAcquisitionManager.executeAcquisition(DefaultAcquisitionManager.java:108)
                                   [       ]   at org.micromanager.acquisition.internal.DefaultAcquisitionManager.runAcquisitionWithSettings(DefaultAcquisitionManager.java:84)
                                   [       ]   at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
                                   [       ]   at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
                                   [       ]   at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
                                   [       ]   at java.lang.reflect.Method.invoke(Method.java:498)
                                   [       ]   at org.micromanager.internal.zmq.ZMQServer.runMethod(ZMQServer.java:270)
                                   [       ]   at org.micromanager.internal.zmq.ZMQServer.parseAndExecuteCommand(ZMQServer.java:325)
                                   [       ]   at org.micromanager.internal.zmq.ZMQServer.lambda$initialize$1(ZMQServer.java:83)
                                   [       ]   at java.util.concurrent.FutureTask.run(FutureTask.java:266)
                                   [       ]   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
                                   [       ]   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
                                   [       ]   at java.lang.Thread.run(Thread.java:748)
```
So this line?

https://github.com/micro-manager/micro-manager/blob/d455aacce58721e8442d99bca9630106732c17b6/mmstudio/src/main/java/org/micromanager/acquisition/internal/MMAcquisition.java#L218

[CoreLog20200729T094422_pid22508.txt](https://github.com/micro-manager/micro-manager/files/4993546/CoreLog20200729T094422_pid22508.txt)

impact27, 2020-07-29:
I guess this was fixed by https://github.com/micro-manager/micro-manager/pull/829. I will try when I can

impact27, 2020-07-29:
Under which conditions could the number of channels be 0? That doesn't really make sense does it? Even if `store_.getSummaryMetadata().getChannelNameList().size();` is `0` would this work?

impact27, 2020-07-30:
Fixed by  #829., I will reopen if the problem pops up again.

closed 2020-07-30 by impact27