micro-manager/micro-manager #891: Unable to acquire images after selecting Camera-2 in MDA

bna080, 2020-07-28:
1. Connect two identical cameras
2. Launch Micromanager 2.0 nightly build July 27, 2020
3. From Hardware Configuration wizard add two cameras and Multi-Camera support from Utilities
4. From Device Property Browser set Multi Camera-Physical Camera 1 to Camera 1 and Multi Camera-Physical Camera 2 to Camera 2
5. In configuration group create a new group and add Core-Multi Camera property to it. Name the group MultiCam.
6. Go to Multi-Dimensional Acquisition
7. Configure e.g. 10 Time points and enable Channels.
8. Under Channels section set Channel group to MultiCam.
9. Enable Channel "Camera-2" and disable all other channels (Camera 1 and Multi Camera).
10. Click "Acquire". 
11. Error message reporting Acquisition failure pops up. This does not happen if Camera 1 or Multi Camera is enabled.

marktsuchida, 2020-07-28:
(I take it you meant the `Core-Camera` property in step 5.)

I can reproduce this with two DemoCamera `DCam`s, provided that the MDA interval is zero (implying a sequence acquisition). From the error log, it appears to be an index computation error in the Clojure acquisition engine code.

This might take some time to fundamentally fix (there appears to be, in the code, some confusion between MDA channel index and Multi Camera channel index, and all uses will need to be examined in order to fix without breaking other behavior), but I will look into it.

A workaround in the meantime is to switch off channels in MDA and set to `Camera 2` in the main window when running an acquisition with `Camera 2` only. (In my testing, that was the only combination that failed.)

bna080, 2020-07-29:
We have tried the workaround with either two physical PVCAM cameras or with two DCam cameras, but unfortunately we are still getting the same error message: Acquisition failed. 
We disabled the Channels group in MDA window and, set the Core camera to DCam2 and clicked "Acquire!" in the MDA window.

marktsuchida, 2020-07-29:
Sorry, you're right -- now that I've checked again it is sensitive to additional state of the MDA channels settings, and also appears to use the wrong camera even in under the combination that I thought was working.

An alternative workaround is to replace your `Core-Camera` config group with one that contains the two properties: `Multi Camera-Physical Camera 1` and `Multi Camera-Physical Camera 2`. Create 3 presets: two that have a single camera as `Physical Camera 1` and `Undefined` for `Physical Camera 2`, and a third that has two physical cameras. Then set `Core-Camera` to always be `Multi Camera` (e.g. using the `System` `Startup` preset or the default camera in the Hardware Configuration Wizard). Make sure to remove the config group containing `Core-Camera`.

The camera can then be selected in the MDA Channels. It's still not perfect: it works for `Camera 1`, `Camera 2`, `Camera 1` + `Camera 2`, and `Multi Camera`; but it produces spurious (empty) channels when other combinations are used (which are hopefully not needed).

nicost, 2022-12-12:
Fixed in #1555

closed 2022-12-12 by nicost