micro-manager/micro-manager #892: Enter key in HCW not handled correctly

marktsuchida, 2020-07-28:
In the Hardware Configuration Wizard, Step 2, add a (non-hub) device. When the dialog with device label and pre-init properties pops up, hit Enter. Expected behavior is same as OK button (confirm the dialog). Actual behavior is that the dialog is confirmed _and_ the same device is added again, popping up the dialog again.

marktsuchida, 2020-09-03:
Might be Windows only -- could not reproduce on macOS.

nicost, 2020-09-03:
Just confirmed this indeed happens as described in the original report on Windows.

marktsuchida, 2020-09-10:
This seems to only happen when the Enter key was used instead of clicking the "Add..." button. There may be further conditions. Reproduced with Utilities' Multi Stage.

marktsuchida, 2020-09-11:
This might be sensitive to something murky (JDK patch version?). We found that at least one local build did not have the issue.

nicost, 2020-09-11:
I see it with AdopOpenJDK 1.8.0_242, as well as with the JDK 1.8.0_152 that ships with MM.