micro-manager/micro-manager #894: Hyperstack ordering in ImageJ

mabruce, 2020-07-31:
Loading images in ImageJ sometimes produces the wrong channel--Z ordering based on how the stack is acquired.

Looking at [StorageMultipageTiff](https://github.com/micro-manager/micro-manager/blob/d455aacce58721e8442d99bca9630106732c17b6/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/StorageMultipageTiff.java#L621) and [ij.io.Opener](https://github.com/imagej/ImageJA/blob/master/src/main/java/ij/io/Opener.java#L814), ImageJ metadata support for proper hyperstack ordering when not loading through Bio-Formats can be partially accomplished. ImageJ appears to default to XYCZT order, and there is support for indicating XYZCT instead if the description has `order=zct`. All other `order=` variants are ignored.

Time in other ifd ordering would require a push to upstream ImageJ and modifying the HyperStackConverter lines in ij.io.Opener. Going by #716 this isn't necessarily supported by Micro-Manager anyway.

marktsuchida, 2020-07-31:
It looks like an [older version of StorageMultipageTiff](https://github.com/micro-manager/micro-manager/blob/9e674a1873c73dd00941c9d6b55ded5800f88b53/mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/StorageMultipageTiff.java#L601) had something along those lines, writing `order=` only in the case of `order=zct`.

@nicost (when you're back from vacation) It looks like you expanded and later commented out this section of code that writes `order=`. Do you see any reason against reverting to the version I linked above?

mabruce, 2020-07-31:
I did a quick test of that change on my system and Fiji appears to load it correctly for a zct acquisition.

nicost, 2020-08-04:
Fixed in #896.  In my previous testing, I never got any of the order tags to work, hence removed them.  @mabruce actually looked at the ImageJ source code and re-figured it out.  Added comments to the code so that the next person (in about 7 years I guess) will figure things out faster.

closed 2020-08-04 by nicost