micro-manager/micro-manager #895: Micromanager Freezes

impact27, 2020-08-03:
Using gamma 20200728, micromanager randomly freezes during acquisition.
While acquireing, the interface freezes, nothing responds. 
20s later, an error window appears informing me that the camer timed out. This window is the only one I can close.

Logs of two crashing instances

[CoreLog20200803T103006_pid25808.txt](https://github.com/micro-manager/micro-manager/files/5016781/CoreLog20200803T103006_pid25808.txt)
[CoreLog20200803T155303_pid16604.txt](https://github.com/micro-manager/micro-manager/files/5016782/CoreLog20200803T155303_pid16604.txt)

nicost, 2020-08-13:
Sorry for the delay.  It will be best to contact Andor about the issue.  It appears that the camera is not sening data, and MM times out waiting for an image to appear.  The Andor device adapter sends the following error:

2020-08-03T15:48:37.185456 tid9116 [IFO,dev:Andor sCMOS Camera] [AcquireFrameInSequence] WaitBuffer returned false, no data!

Andor maintains the device adapter code themselves, so it will be best to confer with them.

impact27, 2020-08-14:
Thanks. I just wonder if it is possible that the causality is reversed. When the crash happens, it looks like the interface freezes first, and 20s later I get the error window. Would you expect the camera not returning a frame to freeze the interface?

nicost, 2020-08-14:
Anything blocking function call that happens on the EDT (the Java Swing thread that runs the UI) will block the UI.  I assume that is what happens here (it is more or less the only way to block the UI, but it is very good at doing so;), but I will check the stack traces in your logs.

closed 2020-11-17 by nicost

impact27, 2020-11-17:
I forgot to come back to close this issue. The problem was solved with https://andor.oxinst.com/assets/uploads/andor-support-resources/FAQ065.pdf
In short: a USB power managment feature was closing the connection to the camera.