micro-manager/micro-manager #909: Error setting Config of single property

ieivanov, 2020-08-19:
I created a Config group containing a single property, the camera frame rate, which is a double. In that case micromanager will allow you to change the property directly. When I call `mmc.setConfig("CameraFramerate", "10")` I get an error that such config group does not exist. Using a double or int for the second doesn't work either. Calling `mmc.setProperty("Camera", "Framerate", 10)` works properly.

nicost, 2020-08-19:
This is expected behavior.  The Micro-Manager UI let's you change the number directly as a convenience, the core does not have that concept, i.e. it only knows about a single group (the name you gave it when you created the group).  Is this a problem/Issue?

ieivanov, 2020-08-19:
Thanks @nicost, it's not really a problem, I just figured it might work. You're welcome to close this issue.

closed 2020-08-19 by nicost