micro-manager/micro-manager #916: Allow plugin classes (and resources) to be accessed from outside

marktsuchida, 2020-08-26:
Currently, each plugin JAR (_not_ each plugin class) is given a separate class loader (whose parent is `MMStudio`'s class loader, which is also the ImageJ1 class loader).

This was done for two reasons:
1. To speed up the plugin loading process by preventing SciJava's `DefaultPluginFinder` from scanning every single JAR on the classpath (this was the initial motivation).
2. To forbid other code from depending on plugin-internal code.

While reason 2. is still valid, it would be nice to (a) be able to quickly test plugin code by invoking it from scripts and (b) be able to have single JARs that contain both library code (APIs intended for external use) and plugins.

Furthermore, the use of individual class loaders complicates reflective code (such as the ZeroMQ server).

I think (hope) that we can ignore any performance concerns, because Fiji launches fast enough these days, at least the second time, and in any case our future should be to use the ImageJ2 launcher.

So **we want to simplify the situation by putting everything on the classpath**. This is in fact closer to how the SciJava plugin mechanism is intended to be used, so it will also help when we move to shipping ImageJ2 instead of ImageJ1.

There is just one problem: currently, when MM is launched by double-clicking its icon, the classpath consists exactly of `ij.jar` (only), because that's how ImageJ1 works. (The classpath is whatever you set it to be if you launch from an IDE, of course.)

So it would appear that we need to
- Migrate to the ImageJ2 launcher (which puts all available JARs on the classpath),
- Move all plugin JARs (from `mmplugins` and `mmautofocus`) into the classpath (`jars` directory), and
- Update our plugin loading code (mostly just delete the JAR enumeration code we have and replace with straightforward calls to SciJava).

This, of course, is a rather large undertaking (although certainly desirable in the long term). Among many other things, it requires that we get all of our dependencies to be at compatible versions with ImageJ2.

A shorter-term, easier fix might be to create our own "main" class loader (as a child of the ImageJ1 one), and load everything (including MMStudio itself) using it. I'm not 100% sure what the consequences of this would be -- I would worry in particular about ImageJ macros accessing MM code.

Unfortunately for us, it is not possible to change either the classpath or the URLs searched by an existing `URLClassLoader` at runtime.

There might be some other way to hack this, so I'll keep thinking, but I'm out of ideas for now. Suggestions would be highly appreciated.

(Mildly related: #331.)

nicost, 2020-08-26:
>     To forbid other code from depending on plugin-internal code.
> While reason 2. is still valid

I agree there s an argument to be made for this to be the case, on the other hand, it creates cumbersome barriers that make it difficult to develop new functionality, and I have run into this limitation quite a few times in the past, but never encountered a situation where this barrier prevented a problem.  So, I would do nothing to prevent this (like we do nothing to prevent plugins calling internal mmstudio code).

In my single (futile) attempt to change the classloader, I tried to load the plugins with the ImageJ classloader (which is used to load the ImageJ Micro-Manager plugin).  I am not sure why that did not work, but it seemed a logical way to go.

B.t.w.:

> This, of course, is a rather large undertaking (although certainly desirable in the long term). 

I tested the ImageJ2 launcher by replacing the existing launcher, and it works without any issues (at least on Windows).  I am probably missing some of the complexities, but this seems straight forward.

marktsuchida, 2020-08-26:
Oops, I wrote "reason 2 is still valid" when reason 2 read "to discourage" and then later edited it to "to forbid" -- so I meant only to say that discouraging such practice is still valid. (Of course one would never encounter a problem that was averted in the first place.) In any case, it should be clear from the rest of my sentence that I fully support removing this barrier, as I agree that the nuisances are way too great to be worth it. I was just documenting the history there, because looking at the code reminded me that performance was the original motivation -- something I had forgotten.

I've also run MM from the ImageJ2 launcher multiple times and usually nothing immediately seems wrong (surely the gold standard of software testing ;). In any case, I don't think we can detect any hidden issues until we make the switch. If we're going this route, I think the launcher migration should be a separate ticket as it can be done entirely before changing how plugins are loaded. I think it is technically straightforward, just a lot of small, boring details to manage.

henrypinkard, 2020-08-31:
I also agree that removing this barrier makes sense. I think it can in fact be a nice feature to be able to intentionally breach this barrier when developing fast prototype code, and then revise to something less brittle after testing.

I don't think the switch to a single ClassLoader will affect the ZMQServer stuff since it just takes in a collection of ClassLoaders at the moment and searches them all, but I can make sure of that once the switch is made

marktsuchida, 2022-02-07:
It might be a good time to tackle this once we merge `vs2019` soon. Presumably a prerequisite for a fully Maven-based Java build/packaging.

nicost, 2023-03-28:
Just ran into this problem again (can not use the ZProjector Executor code through the pycromanager bridge).  It is a bit debilitating, and lots of nice code is not reusable at runtime.  Any paths forward?

henrypinkard, 2023-03-28:
I had thought I'd fixed this for ZMQ bridge by just grabbing all the class loaders and passing them in at its construction. See here:

https://github.com/micro-manager/micro-manager/blob/9eca07ec58cac6d3f76b221f1e75e040b5b751d2/mmstudio/src/main/java/org/micromanager/internal/MMStudio.java#L601

It works for Magellan, so my guess here is that the classloader for ZProjector Executor isn't included in the above code. Easiest potential short term fix would just be to include it

nicost, 2023-03-28:
Good call, the ZPorjector is a DisplayGearMenuPlugin, so not included.  These disparate ClassLoaders have been such a pain.

nicost, 2023-03-28:
Downside of adding all these classloaders is that it takes even longer for the ZMQ server to load. I'll create a PR, but wonder whether it is a good idea to start the ZMQ server on a separate thread.

nicost, 2023-08-29:
Currently, all plugin are loaded in one classloader (#1773), which means that plugins can access to each other.