micro-manager/micro-manager #920: Migrate C++ build to Visual Studio 2015-2019

marktsuchida, 2020-08-28:
Migrate the Windows build from Visual Studio 2010 SP1 and WindowsSDK7.1 to Visual Studio 2015 (v140) or 2019 (v142), with C++ language standard set to C++14 (the VS2019 default) for the time being (because C++17 introduces incompatible changes and should be handled as a second migration). VS2015 through 2019 are very compatible with each other.

The challenge is nothing technical about the code, but mostly about migrating the automated builds and taking care of the handful of modules that need special consideration due to version compatibility issues.

Until we address this, we welcome contributions (pull requests) that address any compiler errors when building individual device adapters with VS2019 (or 2015, 2017). These will be easiest for us to accept and merge if they do not prevent compilation with VS2010 and if they are limited to a single module or topic (see #701 for an example of the latter). Changes that replace lines scattered across many parts of many modules may be challenging to immediately merge, due to our ongoing phase out of Subversion.

henrypinkard, 2020-10-20:
Any chance this could be prioritized to 2.0? It would be really nice to be able to add to c++ layer without having to deal with decade old software (and would probably increase community contributions as well)

marktsuchida, 2020-10-20:
Perhaps, but only if 2.0 is significantly delayed for some other reason (which I really want to avoid). Unlike the other build-related renovations, this doesn't conflict that much with work on 2.0 itself, but the problem is that we still build 1.4 from the same C++ codebase for those very few users that unfortunately still require it, so it's hard to do until after retiring the SVN repo from the 2.0 build. I am definitely not going to allow the C++ codebase to be split into 2 versions.

One interim solution might be to simplify the build of out-of-tree device adapters (including shipping such adapters with our binary installers). Not sure if I'll get to this before or after 2.0, but I need it for a few projects and is probably even better for contributors.

henrypinkard, 2020-10-20:
The latter sounds cool, tag me if you ever put up an example of that

ischoegl, 2020-10-23:
FWIW ... since I did just that over the summer (i.e. build an out-of-tree device adapter), here are steps for a toolchain that worked for me

#### Windows7.1SDK

* Download iso from [here](http://download.microsoft.com/download/F/1/0/F10113F5-B750-4969-A255-274341AC6BCE/GRMSDKX_EN_DVD.iso)
* Unzip, and run the installer within the setup folder (which avoids the check for .NET 4)
* Make sure to have Visual C++ compilers unticked (this is broken and will fail to install, see [post](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/805ab640-a1a5-43af-87cf-8020c16fd32c/sdk-fails-to-install-at-quotcompilers-for-x86quot))
* Apply patch KB2519277 (which applies VS2010 SP1)

#### Visual Studio 2010

* Install Visual C++ 2010 Express
* Upgrade to Visual Studio 2010 SP1

#### Visual Studio Code 2019 Community

* Run installer, check Desktop development with C++

#### Micro-Manager

* Download binaries

#### Compile Driver

* Clone the [micro-manager/micro-manager](micro-manager/micro-manager) repository
* Within `DeviceAdapters/<adapter>`, open `<adapter>.vcxproj`, and compile 
* There may be some installation specific object paths that need to be adjusted, but that's not too difficult

ischoegl, 2020-10-23:
FWIW ... I spent a bit of time coming up with a toolchain to create an out-of-tree device adapter this summer, see this [Gist](https://gist.github.com/ischoegl/d5a2b490b92eee084931553eec597e1f). It's obviously based on the [official instructions](https://micro-manager.org/wiki/Building_Micro-Manager_Device_Adapters), but I found that some steps no longer worked.

nanthony21, 2021-02-16:
I would love to see this completed. Is there anything I can do to help move it forward?

nicost, 2021-02-16:
I think that the main obstacle is the automated build.  We could either try to get this to work through github actions (like pymmcore, which raises the question which compiler(s) are used there), or by setting up a new build machine using Jekyll, but in an environment with a newer VS (I guess that 2019 is the best choice right now).  I can try to carve out some time for the second option, but in the long run it may be great to build using github actions.

nanthony21, 2021-02-16:
I started on a Github Actions file to build Micro-Manager. However, it currently fails at the `ant build` step since it can't find VS2010.  Has anyone been able to successfully build using VS2019?

nicost, 2021-02-17:
I am not sure what the issues are.  @marktsuchida had a PR (#701) that addressed one of the issues.  I am not certain of the status of that PR since it was merged into the github source, whereas we are still working with the svn repo as the primary source for all C++ code.

nanthony21, 2021-02-17:
It seems like I am able to build successfully with with VS2019. However the Ant build script seems to only work with VS2010. So maybe it's just a matter of improving the Ant script?

nicost, 2021-02-17:
What is the error when running ant?

nanthony21, 2021-02-17:
This pops up during the `build-buildtools` phase

```

BUILD FAILED
set-versionstring:
D:\a\micro-manager\micro-manager\build.xml:43: The following error occurred while executing this line:

set-version:

build-cpp:
    [mkdir] Created dir: D:\a\micro-manager\micro-manager\build\Release\x64
D:\a\micro-manager\micro-manager\buildscripts\Ant\windowsprops.xml:143: Cannot locate vsvars32.bat for Visual Studio 2010

Total time: 2 seconds
Error: Process completed with exit code 1.
```

nicost, 2021-02-17:
windowsprops.xml looks explicitly for that file.  I do not know enough about the windows build, but would try to edit that file so that it looks for the build tools that you are using to build.

marktsuchida, 2021-02-17:
Indeed, I remember I hard-coded the VS2010 or WindowsSDK7.1 paths into the Ant script, because that was the only way for that version. The modern (2015 and 2019) versions would probably be simpler thanks to `%VCVARSALL_DIR%`, but making it work with either version might be complicated.

@nicost pymmcore is built with VS2015 (`14.0` in the GitHub Actions script, which appears 3 times and ought to be defined as a variable), but it should be trivial to switch to VS2017 or 2019 once the reasons for using 2015 go away.

Whether it is practical to use GitHub Actions to build MM (esp. in view of `3rdparty`) I am not sure.