micro-manager/micro-manager #923: Migrate to the ImageJ2 launcher and core JARs

marktsuchida, 2020-08-28:
In other words, distribute Micro-Manager packaged with ImageJ2 instead of ImageJ1.

Among other benefits, this will facilitate running MM and Fiji in the same process (although we cannot distribute with Fiji for license compatibility reasons).

Distribution via an ImageJ update site is a separate topic.

Likely prerequisite for #916.

nanthony21, 2021-11-13:
I'm interested in seeing this happen. It would be nice to be able to used ImageJ-Ops which I don't think is currently possible. Any idea what major roadblocks there are? Is there a reason why this should be delayed?

marktsuchida, 2021-11-15:
@nanthony21 Mainly the fact that there is no "plain" ImageJ2 distribution at the moment (see note "Where is ImageJ2?" at https://imagej.net/downloads).

Other than that, there are many details that need to be checked (on each platform), but I'm not aware of any particular roadblock.

nanthony21, 2021-11-15:
Makes sense. Any objection to using FIJI?

marktsuchida, 2021-11-17:
@nanthony21 As mentioned above Fiji is GPL. Micro-Manager is not GPL-compatible, so we cannot offer a download that contains both. Also download would become large.

nanthony21, 2021-11-17:
Ok, makes sense.