micro-manager/micro-manager #925: Image viewer zoom-in button causes window to expand beyond screen

marktsuchida, 2020-08-28:
I believe this is a regression; previously the zoom-in button would cause the Window to expand as far as it can while fitting in the screen (i.e. with its bottom and/or right edge up against the screen edges).

Observed on Windows.

@nicost Was this an intentional change by any chance?

marktsuchida, 2020-08-28:
I should add that this was observed with dual monitors.

nicost, 2020-08-28:
With a single monitor, it expands to the size of the monitor.  Buttons can be hidden, but the window can be dragged to the top left corner and will fit the screen.  I don't remember the full history, but do remember that the top left corner of the window would previously move all by itself  when zooming in(there was a ticket about this).  I like that is no longer the case, and repeated clicking at the same position now works as intended.

marktsuchida, 2020-08-28:
Agree that the top-left should not move; that has always been the behavior in the gamma branch.
So I guess this is a multi-monitor-specific issue.

marktsuchida, 2020-09-01:
Tested on macOS with single monitor. The window does size itself, but not always correctly: it stays too wide to fit in the screen and is also too tall on the first zoom.

closed 2020-11-26 by marktsuchida