micro-manager/micro-manager #927: Some device adapters fail to load / unavailable

calocedrus, 2020-09-01:
Hello,

These are my very first steps into micro-manager which I'd like to use with ASI's ASITiger device. 
My OS is linux ubuntu 18.04, and I've managed to successfully build MM2 (from the master branch).
I've built MM as an imageJ plugin (imageJ was installed via FIJI).

All seems to build correctly.

I can start MM (from within the FIJI directory), but when trying to configure the device (tools > Hardware Configuration Wizard) I notice that 4 devices are unavailable, as shown below.
[image]

Since I'm interested in the ASITiger device manager, I've traced any error related to this device. The only one found is in the terminal log (from where I started MM), when I open the Hardware Configuration Wizard, it shows the following:

```
2020-09-01T11:06:18.042743 tid139965953734400 [IFO,App] ASITiger
2020-09-01T11:06:18.075066 tid139965953734400 [IFO,App] Error: Unable to load ASITiger library: Failed to load device adapter "ASITiger" [ Failed to load module "~/Programs/fiji_imageJ/Fiji_app/libmmgr_dal_ASITiger.so.0" [ ~/Programs/fiji_imageJ/Fiji_app/libmmgr_dal_ASITiger.so.0: undefined symbol: _ZN4CDACC1EPKc ] ]

```
The corresponding module, found in ~/Programs/fiji_imageJ/Fiji_app/libmmgr_dal_ASITiger.so.0, was built without any error, this is the corresponding section during make:

```
mv -f .deps/ASITigerComm.Tpo .deps/ASITigerComm.Plo
mv -f .deps/ASIScanner.Tpo .deps/ASIScanner.Plo
mv -f .deps/ASIZStage.Tpo .deps/ASIZStage.Plo
mv -f .deps/ASIXYStage.Tpo .deps/ASIXYStage.Plo
/bin/bash ../libtool  --tag=CXX   --mode=link g++ -I~/Programs/micro_manager/micro-manager/DeviceAdapters/../MMDevice -pthread -I/usr/include -g -O2 -module -avoid-version -shrext ".so.0"  -o libmmgr_dal_ASITiger.la -rpath ~/Programs/fiji_imageJ/Fiji_app ASICRISP.lo ASIClocked.lo ASIFWheel.lo ASIHub.lo ASILED.lo ASILens.lo ASIPLogic.lo ASIPiezo.lo ASIPmt.lo ASIScanner.lo ASITiger.lo ASITigerComm.lo ASIXYStage.lo ASIZStage.lo ~/Programs/micro_manager/micro-manager/DeviceAdapters/../MMDevice/libMMDevice.la 
libtool: link: g++  -fPIC -DPIC -shared -nostdlib /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/7/crtbeginS.o  .libs/ASICRISP.o .libs/ASIClocked.o .libs/ASIFWheel.o .libs/ASIHub.o .libs/ASILED.o .libs/ASILens.o .libs/ASIPLogic.o .libs/ASIPiezo.o .libs/ASIPmt.o .libs/ASIScanner.o .libs/ASITiger.o .libs/ASITigerComm.o .libs/ASIXYStage.o .libs/ASIZStage.o  -Wl,--whole-archive ~/Programs/micro_manager/micro-manager/DeviceAdapters/../MMDevice/.libs/libMMDevice.a -Wl,--no-whole-archive  -ldl -L/usr/lib/gcc/x86_64-linux-gnu/7 -L/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/7/../../.. -lstdc++ -lm -lc -lgcc_s /usr/lib/gcc/x86_64-linux-gnu/7/crtendS.o /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/crtn.o  -pthread -g -O2   -pthread -Wl,-soname -Wl,libmmgr_dal_ASITiger.so.0 -o .libs/libmmgr_dal_ASITiger.so.0
libtool: link: ( cd ".libs" && rm -f "libmmgr_dal_ASITiger.la" && ln -s "../libmmgr_dal_ASITiger.la" "libmmgr_dal_ASITiger.la" )
make[2]: Leaving directory '~/Programs/micro_manager/micro-manager/DeviceAdapters/ASITiger'

```
And during `make install`:
```
Making install in ASITiger
make[2]: Entering directory '~/Programs/micro_manager/micro-manager/DeviceAdapters/ASITiger'
make[3]: Entering directory '~/Programs/micro_manager/micro-manager/DeviceAdapters/ASITiger'
make[3]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '~/Programs/fiji_imageJ/Fiji_app'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   libmmgr_dal_ASITiger.la '~/Programs/fiji_imageJ/Fiji_app'
libtool: install: /usr/bin/install -c .libs/libmmgr_dal_ASITiger.so.0 ~/Programs/fiji_imageJ/Fiji_app/libmmgr_dal_ASITiger.so.0
libtool: install: /usr/bin/install -c .libs/libmmgr_dal_ASITiger.lai ~/Programs/fiji_imageJ/Fiji_app/libmmgr_dal_ASITiger.la
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/sbin" ldconfig -n ~/Programs/fiji_imageJ/Fiji_app
----------------------------------------------------------------------
Libraries have been installed in:
   ~/Programs/fiji_imageJ/Fiji_app

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
make[3]: Leaving directory '~/Programs/micro_manager/micro-manager/DeviceAdapters/ASITiger'
make[2]: Leaving directory '~/Programs/micro_manager/micro-manager/DeviceAdapters/ASITiger'
```
So from the above excerpt, it seems ASITiger is built successfully.
For information, the other device manager not loading shows:

```
2020-09-01T11:31:03.042323 tid140543609165568 [IFO,App] ThorlabsDCxxxx
2020-09-01T11:31:03.043823 tid140543609165568 [IFO,App] Error: Unable to load ThorlabsDCxxxx library: Failed to load device adapter "ThorlabsDCxxxx" [ Failed to load module "~/Programs/fiji_imageJ/Fiji_app/libmmgr_dal_ThorlabsDCxxxx.so.0" [ ~/Programs/fiji_imageJ/Fiji_app/libmmgr_dal_ThorlabsDCxxxx.so.0: undefined symbol: TLDC2200_getPWMDutyCycle ] ]

```
And while the SerialManager and USBManager are said to be unavailable, they don't return any error as the above 2 device managers do, the log from the terminal shows (also when I open the Hardware Configuration Wizard):

```
2020-09-01T11:31:03.105703 tid140543609165568 [IFO,App] SerialManager
[...]
2020-09-01T11:31:02.894138 tid140543609165568 [IFO,App] USBManager

```

All the other ASI device managers are available. 
Should I further investigate ASITIger issue with the author of the device adapter or is the .so library unable to load a problem related to MM2, my building it and/or linux (ubuntu 18.04)?

jondaniels, 2020-09-01:
I think the crux is to figure out why the device adapter compiles successfully but does not load due to "undefined symbol: _ZN4CDACC1EPKc".  

Googling similar error messages suggests a dependency might be out of date.  But the ASITiger device adapter shouldn't depend on anything.  It does use serial communication so maybe the SerialManager unavailability is related.  However, there shouldn't be a difference between ASIStage and ASITiger in that respect.

I can confirm that the 1.4 branch compiles and works fine on Win64.  I'm fuzzy on how device adapters make it to the MM2 master branch.

Any ideas @nicost and/or @marktsuchida?

nicost, 2020-09-01:
They also work fine under win64 in the MM2 branch (the Core and device adapters are identical between 1.4 and 2.0).  

I noted that the ASIDac cpp and header files were not listed in the Makefile.am.  That could lead to missing symbols.  I added them to the subversion repo, that change will make its way to github soon.  

You will also need the SerialManager. It is possible that it did not find any serial ports on your computer. If I understand the code correctly, it looks for files like /dev/ttyS, /dev/ttyUSB, /dev/ttyACM.  How are the serial ports on your computer named?

marktsuchida, 2020-09-01:
I think Jon's suggestion is the most likely. Did your `libmmgr_dal_ASITiger.so` and `libMMCoreJ_wrap.so` come from the same build (same invocation of `make`)? If not, and there is any chance that something changed in between (such as installing or updating system libraries or compilers), you might need to rebuild everything after a `make clean`.

calocedrus, 2020-09-02:
Thanks to all of you.
@marktsuchida yes all came from a same build (it was a total rebuild, including `./configure`).

@nicost

> I noted that the ASIDac cpp and header files were not listed in the Makefile.am

You nailed it: I've added them, the .so (dll) loads correctly when I open the HW configuration Wizard and ASITiger becomes available!
For the SerialManager, I believe the reason why it wasn't available is because I installed MM in a local directory under `/home` and didn't run MM as root.
Starting MM with `sudo ./mmimagej` makes the SerialManager available.
That's 2 problems solved.

I don't use nor am I planning to use the ThorlabsDCxxxx manager, but I've tried to debug the issue: the undefined symbol `TLDC2200_getPWMDutyCycle` can be found in `DeviceAdapters/ThorlabsDCxxxx/include/TLDC2200.h`; and this header wasn't in the Makefile.am, adding it as
[...]
libmmgr_dal_ThorlabsDCxxxx_la_SOURCES =  \
                          include/TLDC2200.h \
[...]
and rebuilding all after a `make clean` didn't help. 
The USBManager remains unavailable but reading the associated documentation, I may not need it. This issue is almost closed then.

nicost, 2020-09-02:
The Makefile.am change is now in the github master branch.  

As for the ThorlabsDCxxxx adapter, I am afraid that it has dependencies on Windows-only dlls.   The header file TLDC2200 is already included at compile time through an include in DC2200.cpp, however, the linker does not add in the symbols that are defined in that header file.  Funny thing is that the linker on Windows will complain about this, but apparently, gcc/clang does not and happily builds a library that does not resolve all symbols.  I think that the simplest solution is to remove the ThorlabsDCxxxx from DeviceAdapters/Makefile.am.  If anyone needs this functionality on Mac/Linux, then we'll need the binaries (or souce code) from Thorlabs.

closed 2020-09-02 by nicost