micro-manager/micro-manager #950: Crop to ROI behavior is broken with multi-camera utility

shahlab-ucla, 2020-09-17:
Drawing a rectangle ROI and clicking the crop to ROI button on an image snapped using multi-camera returns an error stating that no ROI was found.

This works as expected with individual cameras.

If a frame is snapped with the multi-camera and then transferred to ImageJ, drawing an ROI and then clicking the crop to ROI button works as it should.

In cases where the the on-the-fly processor pipeline is used, it would be nice if the crop to ROI took any configured transformations into consideration so that the correct region of each sensor is cropped on each camera. The crop seems to occur before images go into the processor pipeline (presumably because cropping is handled by the camera driver?), which results in mismatched ROI's if an off-center one is used when one camera is being rotated and/or mirrored to correspond to the other.

This can be reproduced by adding a second demo camera and the multi camera utility to the demo configuration (see attached).
[MMConfig_demo.cfg.txt](https://github.com/micro-manager/micro-manager/files/5241734/MMConfig_demo.cfg.txt)

marktsuchida, 2020-09-18:
I can reproduce this. The error message is
```
Error: There is no selection in the image window.
Use the ImageJ rectangle tool to draw the ROI.
```
despite there being a rectangular selection.

marktsuchida, 2020-09-18:
The "set ROI to center quad" and "clear ROI" buttons work with multi-camera, so the problem is probably with getting the selection from the GUI.

Until this gets fixed, a temporary workaround is to set the ROI from the Script Panel:
```
mmc.setROI(128, 128, 256, 256);    // x, y, width, height
```

shahlab-ucla, 2020-09-22:
Thanks Mark. Is it possible to crop to ROI from a script as well? I poked around the API but wasn't able to find this.

marktsuchida, 2020-09-22:
@shahlab-ucla Micro-Manager datasets do not generally allow editing. You would have to either create a new, cropped datastore, or first export to ImageJ.

nicost, 2022-07-17:
Fixed in PR #1476.  Sorry this took so long!

closed 2022-07-17 by nicost