micro-manager/micro-manager #953: Crash when adding Aladdin Syringe Pump to Hardware Configuration

zhangxiaosa, 2020-09-19:
Click Devices->Hardware Configuration Wizard. Then click Next.
In step2, select Aladdin Syringe Pump and click Add. Then click Ok, the program will crash.
Here are the log and error messages:

```
....
2020-09-19T10:02:43.818659 tid10272 [dbg,Core:dev:Aladdin] Will set property "Port" to ""
2020-09-19T10:02:43.818659 tid10272 [dbg,Core:dev:Aladdin] Did set property "Port" to ""
2020-09-19T10:02:43.818659 tid10272 [IFO,Core] Will initialize device Aladdin
2020-09-19T10:02:43.818659 tid10272 [IFO,dev:Aladdin] AladdinController::Initialize()
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_UNCAUGHT_CXX_EXCEPTION (0xe06d7363) at pc=0x00007ffaf8e03b29, pid=4208, tid=0x0000000000002820
#
# JRE version: Java(TM) SE Runtime Environment (8.0_261-b12) (build 1.8.0_261-b12)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.261-b12 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [KERNELBASE.dll+0x43b29]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\hustzmx\Desktop\RA\nightly\Micro-Manager-2.0gamma\hs_err_pid4208.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

[error occurred during error reporting , id 0xc0000005]

Process finished with exit code 1
```

marktsuchida, 2020-09-21:
I can reproduce this. It crashes even if a serial port is assigned (without the device being connected), although in that case it makes it a little further:
```
2020-09-21T11:17:08.802910 tid39548 [IFO,Core] Will initialize device Aladdin
2020-09-21T11:17:08.802910 tid39548 [IFO,dev:Aladdin] AladdinController::Initialize()
2020-09-21T11:17:08.803857 tid39548 [dbg,dev:COM1] SetCommand -> RAT\r
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_UNCAUGHT_CXX_EXCEPTION (0xe06d7363) at pc=0x00007ff969a43e49, pid=31320, tid=0x0000000000009a7c
#
# JRE version: OpenJDK Runtime Environment (8.0_265-b01) (build 1.8.0_265-b01)
# Java VM: OpenJDK 64-Bit Server VM (25.265-b01 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [KERNELBASE.dll+0x23e49]
```

It looks like missing error handling in the Aladdin device adapter.

nanthony21, 2021-02-12:
It looks like a fix to this ended up getting added through SVN. Can we close this issue?

closed 2021-02-12 by nicost