micro-manager/micro-manager #955: Affine vs linear transformations

marktsuchida, 2020-09-21:
A 2D affine transformation requires 6 parameters and covers rotation, scaling, shearing, and translation. A 2D linear transformation requires 4 parameters and covers rotation, scaling, and shearing only.

So I'm pretty sure the "affine transform" in the pixel size config editor (and possibly other places?) should actually read "linear transformation", or, more colloquially, just "transformation matrix". In other places we do use real affine transformations.

In the pixel size configs, I believe we do use affine transformations internally, which seems like a good idea because there may be different translations between different objectives. So perhaps we should allow setting a translation rather than rewording.

nicost, 2020-09-22:
We actually use a real affine transformation everywhere, however, for the pixel size calibration, the last x1y3 and x2y3 are irrelevant because they specify translation, and they are not exposed there.  It would be very confusing to call it a linear transform, because it isn't. 

We could expose those, however, there is no easy way to measure offsets between objectives (it isn't even obvious which objective is the gold standard to which the others should conform).  Making those number editable in the dialog would make it even more confusing for naive users (I simplified things quite a bit with lots of advice from @henrypinkard, and at this point in time would not start changing things unless there is a very clear plan and strong rational.  Close?

henrypinkard, 2020-09-22:
Definitely want to include the full affine under the hood for users that know what they're doing and can use translations. Seems like it would be confusing to call the same thing different things in different places. Maybe it could make sense to add in the translation column to pixel size calibrator, but keep it grayed out and disabled

marktsuchida, 2020-09-22:
I'm not in a hurry to change this, I feel it's a minor issue, and agree that any change should be carefully considered.

I guess the real problems that I see are: (1) it is not clear (to the user) what happens if you have a non-zero translation set up, but then edit it using the pixel size calibrator and (2) for somebody who is new to MM but familiar with affine transformations, the 2x2 matrix doesn't make sense; such a person would probably conclude that it's a typo for linear transformation, because that is what the GUI appears to be displaying.

For (2), we can kind of side-step the problem by calling it a "transformation matrix" without mentioning "affine" until/unless we decide to display the translation parameters.

For (1), I'm not sure there is any clean solution other than to allow editing the translation. Disabling the editing of displayed translation parameters doesn't make sense to me -- as a user I would wonder why I'm forbidden from editing and be frustrated by the lack of control -- although it is arguably better than not being aware of hidden state. On the other hand, it can be argued that this will not be a big issue as long as non-zero translations are not widely used (does Magellan use them?).

henrypinkard, 2020-09-22:
Magellan does not use them, since unlike xy stages, the zero in pixel coordinate space is arbitrary, thus translations between the two are meaningless. I would think translations only matter if you have multiple cameras (i.e. translation of the pixel coordinates of one relative to the other). I don't think the way it is now is especially confusing. Perhaps it is worth closing and leaving as is until someone using it actually says that they are confused

marktsuchida, 2020-09-23:
Why should we close this? We can leave it open because we don't seem to immediately know what the best solution is, but I don't understand why we should keep terminology that is clearly incorrect.

nicost, 2020-09-23:
I disagree that the terminology is incorrect.  It is an affine transform, just does not expose the translation components. If you really think it is helpful, I am fine showing the translation component un-editable (and let's throw in the constant lower row as well;). Alternatively, we can call the editable numbers "Scale-X", "Scale-Y", "Shear-X", and "Shear-Y".  We could even let the user edit the translation component, just make sure that the numbers of the automated measurement are rejected, since they can have dangerous consequences.

I don't think that it is useful to leave tickets open with something we do not like, but that we do not have a solution for.  It clutters the list of actionable issues.  Let's make a decision what we want this to look like, do it, and close.

marktsuchida, 2020-09-24:
I'm not exactly sure why our views are so different (or why this is such a sensitive(?) issue), but I find the current situation confusing enough to be a problem. Well, not to me, because I can read the source code -- but if I saw this in an unfamiliar program I would worry (incorrectly in this case) that the developers didn't understand the difference between linear and affine transformations, and that would have a negative effect on my confidence with such software. More importantly, people who are less familiar with these concepts might learn them from Micro-Manager, and they could learn incorrectly. That seems like a disservice to the community.

The underlying implementation is indeed an affine transformation, but the GUI only allows editing the linear transformation part of it, so the inaccuracy is real.

Don't get me wrong: I'm not saying that this is a particularly crucial or urgent issue (as hopefully I've made clear). But it is one that I wouldn't want to lose track of.

Finally, @nicost, I'm very confused by your last statement. Should I keep a secret list of every issue I encounter and only post it here when I have a solution to propose? We have many other tickets that I don't immediately know how (best) to solve (both those filed by me and by others), but surely those shouldn't just be closed. I'm even more confused because we all did propose a few (at least partial) solutions above, and it doesn't seem that they have all been rejected (so why conclude that we have no solution?). If clutter is the only issue, we now have milestones to manage that (and their use can be further improved if necessary).

marktsuchida, 2020-09-24:
Getting back to solutions, I think the version of @nicost's last proposal of allowing all 6 parameters to be edited, with the necessary checks/constraints, solves the problem. But I also realized that I'm not sure what motivates the idea of making the translation uneditable, so please feel free to clarify.

henrypinkard, 2020-09-24:
i dont know of any actual code that uses translation. Magellan automatically throws it out. But as you mention its useful for some applications. So it might just add confusion to have extra parameters that dont seem to affect the things people are actually using. But it definitely should be in the core so advanced/customized things could make use if they need. But the bottom row (i.e.entries 7-9) should definitely not be shown because they are always 0 and not even stored by the affinetransform class in Java

nicost, 2020-09-24:
The config file stores all 6 numbers of the affine transform, the Core setPixelSizeAffine function will only accept vectors with 6 doubles.  The reason not show the translation part of the affine transform was exactly to reduce complexity for the user.  Advanced users can always set these values through a script, or by directly editing the config file.  

So, that mainly leaves people who know about 2D affine transforms and are confused to only see 4 values. It is fine with me to add an extra column in that table (although it feels a bit esoteric), however, the automated methods will never enter anything but zero in that column (in fact, I had some bad experiences with crazy numbers inserted there by the automated calibration routine), and I expect very people (if any) to ever enter something useful there. So Mark, if you really think that is better, I'll add an extra column with editable zeros.

marktsuchida, 2020-09-25:
@nicost @henrypinkard Thanks. Sorry for the long post that follows, but there is another aspect of this problem that I haven't mentioned, partly because I was initially hoping to defer till much later thinking through this issue, and partly because the situation is becoming clearer to me now.

The real goal, of course, is to make it the least confusing to users (whether or not they are familiar with linear algebra) -- not just in terms of GUI presentation, but also behavior of the code that _uses_ the affine transformations. For that, we need the transformation matrices to be treated in a uniform and predictable way. The kind of thing that keeps me up at night (much more so than mere terminology issues) is the possibility that we end up in a world where we see inconsistent (and incompatible) behavior between different parts of MM -- we already have enough of those. This particular one may not be a major one, but if we leave around lots of these "mostly correct" things, we will suffer a slow death by a thousand cuts. (Or, more positively, the magic of software is that it allows us to come up with well-defined models and subsequently rely on their behavior, so what's the point if we don't take advantage of that?)

We do not have any such issues right now with these matrices (as far as I am aware) because no code is actually making use of (i.e., applying to data) the translation part (am I correct?). But if we leave it this way (keeping 6 parameters but only ever using 4 parameters), I can almost guarantee that in a few years we will find in our code several places where the translation parameters are discarded, silently overwritten to zero, or even assumed to be zero. (I can't even trust myself not to add such code when it's the easiest way to fix a given problem.) Each such piece of code may work fine by itself, but we could run into situations where (for example) two plugins cannot work from the same config file, or some problem along those lines.

One way to prevent such a situation is to make it a clear rule (A) that if you use the pixel affine transformations (i.e., if you store or read the affine matrices to/from MMCore), then you must always use all 6 parameters correctly. This would mean that (1) the automated pixel calibrator should make sure to write sensible values (i.e., zeros) for translation (unless it gains the ability to measure translation); (2) the pixel calibration editor should, at the very least, be able to _preserve_ nonzero translations, and ideally it provides a simple interface to edit them as well; and (3) users of the affine transformations (such as Magellan) should not make any assumptions about the translation parameters, and apply the full transformation, including translation, under most circumstances(*).

(*) -- translation may be ignored when doing computations on position-less orientations or when comparing scales, of course.

If these rules (A) seem acceptable (see below for alternatives), then I feel that a GUI showing all 6 parameters, editable, is the easiest to understand. Among other things, consistently seeing 6 parameters displayed will encourage future programmers to follow the rules.

In that case, we should mark all uses of the affine matrices that do not handle translation correctly (and, of course, any code that writes garbage to the translation) as bugs, and work toward fixing them.

If, for any reason, it is shown that rules (A) are too strict (for user interface or developer convenience?), then I would say we need some other set of clear rules. Possibilities include (B) disallowing non-zero translation in all cases, at least until we have a use for them (extreme but simple) and (C) making MMCore keep track of whether translation is set or not (i.e. distinguishing between zero translation and null translation). In the case of (C), the pixel calibration editor could be left with 4 parameters only, but should show a warning (_before_ opening) that the existing transformation will be reset to default, or else refuse to edit. (But then, showing six parameters is probably simpler even in this case.)

Among the 3 possible sets of rules (I'm open to others), I prefer (A) for its simplicity, but would like to hear about any cases where it may be a problem. (B) is even simpler and would be an acknowledgment of current practice, but if we adopt it now, we would have to migrate to (C) (not (A)) in the future when (re)allowing the use of translation parameters. And (C) is the most complicated.

marktsuchida, 2020-09-28:
@nicost and I discussed this on the phone the other day, and we agreed that it will be good to consistently use the 6-parameter affine transformations. For the pixel calibration editor GUI, we would like to replace the 2x2 `JTable` with 6 separate `JTextField`s, and display the full 3x3 matrix (with the `[0 0 1]` row at the bottom).

Other places that use the affine parameters, especially those that store them, should be inspected to make sure the translation is handled correctly.

henrypinkard, 2020-09-28:
sounds good to me

closed 2020-10-20 by marktsuchida