micro-manager/micro-manager #960: Magellan: Fails to open dataset acquired on another system

nicost, 2020-09-23:
@henrypinkard: 

Trying to open a dataset acquired in Micro-Magellan on a microscope computer on my laptop. Select "Open Dataset" and select the dataset.  Many progress bars show up and disappear.  The Magellan viewer window shows with a black pane on the left, and no histograms on the right.  There appears to be a dead lock, as the UI no longer responds (have to kill Micro-Manager).

Following is produced by the EDT monitor in MM:

2020-09-22T20:01:04.276648 tid29668 [IFO,App] EDTHangLogger: Missed heartbeat; waiting to see if we are stuck on a single event
2020-09-22T20:01:05.285201 tid29668 [IFO,App] EDTHangLogger: Event handling has exceeded at least 1006 ms (currently 5508 ms since heartbeat timebase)
                                    [       ] Stack traces follow (note: thread states queried later than stack traces)
                                    [       ] Thread 2 [Reference Handler] WAITING
                                    [       ]   at java.lang.Object.wait(Native Method)
                                    [       ]   at java.lang.Object.wait(Object.java:502)
                                    [       ]   at java.lang.ref.Reference.tryHandlePending(Reference.java:191)
                                    [       ]   at java.lang.ref.Reference$ReferenceHandler.run(Reference.java:153)
                                    [       ] Thread 3 [Finalizer] WAITING
                                    [       ]   at java.lang.Object.wait(Native Method)
                                    [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:144)
                                    [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:165)
                                    [       ]   at java.lang.ref.Finalizer$FinalizerThread.run(Finalizer.java:216)
                                    [       ] Thread 4 [Signal Dispatcher] RUNNABLE
                                    [       ] Thread 5 [Attach Listener] RUNNABLE
                                    [       ] Thread 15 [RMI TCP Accept-0] RUNNABLE
                                    [       ]   at java.net.DualStackPlainSocketImpl.accept0(Native Method)
                                    [       ]   at java.net.DualStackPlainSocketImpl.socketAccept(DualStackPlainSocketImpl.java:131)
                                    [       ]   at java.net.AbstractPlainSocketImpl.accept(AbstractPlainSocketImpl.java:409)
                                    [       ]   at java.net.PlainSocketImpl.accept(PlainSocketImpl.java:199)
                                    [       ]   at java.net.ServerSocket.implAccept(ServerSocket.java:560)
                                    [       ]   at java.net.ServerSocket.accept(ServerSocket.java:528)
                                    [       ]   at sun.rmi.transport.tcp.TCPTransport$AcceptLoop.executeAcceptLoop(TCPTransport.java:405)
                                    [       ]   at sun.rmi.transport.tcp.TCPTransport$AcceptLoop.run(TCPTransport.java:377)
                                    [       ]   at java.lang.Thread.run(Thread.java:748)
                                    [       ] Thread 16 [RMI Reaper] WAITING
                                    [       ]   at java.lang.Object.wait(Native Method)
                                    [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:144)
                                    [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:165)
                                    [       ]   at sun.rmi.transport.ObjectTable$Reaper.run(ObjectTable.java:351)
                                    [       ]   at java.lang.Thread.run(Thread.java:748)
                                    [       ] Thread 17 [GC Daemon] TIMED_WAITING
                                    [       ]   at java.lang.Object.wait(Native Method)
                                    [       ]   at sun.misc.GC$Daemon.run(GC.java:117)
                                    [       ] Thread 18 [Java2D Disposer] WAITING
                                    [       ]   at java.lang.Object.wait(Native Method)
                                    [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:144)
                                    [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:165)
                                    [       ]   at sun.java2d.Disposer.run(Disposer.java:148)
                                    [       ]   at java.lang.Thread.run(Thread.java:748)
                                    [       ] Thread 19 [AWT-Shutdown] WAITING
                                    [       ]   at java.lang.Object.wait(Native Method)
                                    [       ]   at java.lang.Object.wait(Object.java:502)
                                    [       ]   at sun.awt.AWTAutoShutdown.run(AWTAutoShutdown.java:295)
                                    [       ]   at java.lang.Thread.run(Thread.java:748)
                                    [       ] Thread 20 [AWT-Windows] RUNNABLE
                                    [       ]   at sun.awt.windows.WToolkit.eventLoop(Native Method)
                                    [       ]   at sun.awt.windows.WToolkit.run(WToolkit.java:316)
                                    [       ]   at java.lang.Thread.run(Thread.java:748)
                                    [       ] Thread 22 [AWT-EventQueue-0] RUNNABLE
                                    [       ]   at sun.java2d.pisces.Renderer.lineTo(Renderer.java:382)
                                    [       ]   at sun.java2d.pisces.Stroker$PolyStack.pop(Stroker.java:1194)
                                    [       ]   at sun.java2d.pisces.Stroker.emitReverse(Stroker.java:415)
                                    [       ]   at sun.java2d.pisces.Stroker.finish(Stroker.java:438)
                                    [       ]   at sun.java2d.pisces.Stroker.moveTo(Stroker.java:348)
                                    [       ]   at sun.java2d.pisces.Dasher.goTo(Dasher.java:155)
                                    [       ]   at sun.java2d.pisces.Dasher.lineTo(Dasher.java:210)
                                    [       ]   at sun.java2d.pipe.RenderingEngine.feedConsumer(RenderingEngine.java:366)
                                    [       ]   at sun.java2d.pisces.PiscesRenderingEngine.pathTo(PiscesRenderingEngine.java:484)
                                    [       ]   at sun.java2d.pisces.PiscesRenderingEngine.strokeTo(PiscesRenderingEngine.java:363)
                                    [       ]   at sun.java2d.pisces.PiscesRenderingEngine.strokeTo(PiscesRenderingEngine.java:163)
                                    [       ]   at sun.java2d.pisces.PiscesRenderingEngine.getAATileGenerator(PiscesRenderingEngine.java:562)
                                    [       ]   at sun.java2d.pipe.AAShapePipe.renderPath(AAShapePipe.java:152)
                                    [       ]   at sun.java2d.pipe.AAShapePipe.draw(AAShapePipe.java:64)
[... 159 lines cut ...]
                                    [       ]   at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2039)
                                    [       ]   at java.util.concurrent.LinkedBlockingQueue.take(LinkedBlockingQueue.java:442)
                                    [       ]   at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1074)
                                    [       ]   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1134)
                                    [       ]   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
                                    [       ]   at sun.awt.shell.Win32ShellFolderManager2$ComInvoker$3.run(Win32ShellFolderManager2.java:586)
                                    [       ]   at java.lang.Thread.run(Thread.java:748)
                                    [       ] Thread 70 [Display calculation executor] WAITING
                                    [       ]   at sun.misc.Unsafe.park(Native Method)
                                    [       ]   at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
                                    [       ]   at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2039)
[... cut ...]
                                    [       ]   at java.util.concurrent.LinkedBlockingQueue.take(LinkedBlockingQueue.java:442)
                                    [       ]   at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1074)
                                    [       ]   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1134)
                                    [       ]   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
                                    [       ]   at java.lang.Thread.run(Thread.java:748)

henrypinkard, 2020-09-23:
I cant reproduce on either mac or windows. Was this using latest nightly for acquisition and opening?

nicost, 2020-09-23:
Good call!  My "reader" was a current source code running under IntelliJ, but did not have the latest dependencies.  After updating it works (although zooming is quite sluggish).  Running under IntelliJ fails to find the NDViewer class, but that is an IntelliJ configuration problem..  Thanks!

closed 2020-09-23 by nicost

nicost, 2020-09-23:
Interesting.  After updating IntelliJ so that it uses the latest of all MMs dependencies, I got the same crash.  Realized that I am using a relatively new OpenJDK 1.8.242.  IntelliJ lets me choose JRE, and when I run it with 1.8.131, I do not get the deadlock, and everything works fine.  Although I did not check, I am quite sure that you can trigger the error by exchanging the JRE from whatever MM uses (1.8.152) to a newer JRE (especially since there are significant improvements to garbage collection).  Since MM will likely upgrade to a newer JRE, it is worth figuring out and fixing this deadlock. I guess that the issue is in NDViewer.  Should I open a ticket there?

henrypinkard, 2020-09-23:
Oh interesting. Yes, please do.