micro-manager/micro-manager #965: MDA Window LUT color is not updated after using Color Picker.  Color in MDA itself is correct.

nicost, 2020-09-30:
Reported by Matt Pearson:

When assigning the LUT to be used on a channel preset in the MDA window the colour shown under the colour column is not what has been set.  For DAPI for example i set 255 blue, 0 green, 0 red, for green 0 blue, 255 green, 0 red and in the pick a color menu it lets me set that but when exiting that menu the color columns shows the wring color. Confusingly the actually LUT applied to the acquired images is correct.  So its merely the displayed colours in the MDA window that are wrong.

marktsuchida, 2020-10-05:
Steps to reproduce:
- Start MM with a newly created user profile (just in case)
- Open MDA; check Channels (only)
- Select a channel group (e.g. `Channel` in the demo config)
- Add two channels (e.g. `Cy5` and `DAPI`)
- Click on the Color column for `DAPI`; select the RGB tab, and enter `0, 0, 255` for red/green/blue
- Click OK to close the color picker
- The color in the table should be updated to pure blue, but is not (the bug)
- Closing and reopening MDA or MM does not update the color display
- I cannot reproduce the part where the selected color gets applied to the image -- I just see white for every channel in the fresh profile

nicost, 2020-10-05:
> I cannot reproduce the part where the selected color gets applied to the image -- I just see white for every channel in the fresh profile

That suggests that colors created before the bug was introduced are retained (in the profile).  Very likely this also happened during the SequenceSettings refactoring.  Hopefully easy to find.

closed 2020-10-08 by marktsuchida