micro-manager/micro-manager #979: Hang on Snap! with new hardware

wwarriner, 2020-10-14:
MMStudio Version 2.0.0-gamma1 20201013 nightly (and earlier versions) hangs when snapping an image using the "Snap" button or "Live" while using a custom camera hardware adapter. However, the "Image -> Album" button correctly displays the expected test pattern. While hanging, CPU utilization according to the Task Manager is approximately the percentage expected for 1 core at 100% usage.

Initial GUI state:
[image]

After clicking image -> album:
[image]

After clicking "Live", the interface is completely unresponsive and no image is displayed.
[image]

The core log for a "Snap" is attached.
[CoreLog20201014T154610_pid8596.txt](https://github.com/micro-manager/micro-manager/files/5380791/CoreLog20201014T154610_pid8596.txt)

The core log using "Live" shows a similar error message, and then continues acquiring images until the process is manually terminated.
[CoreLog20201014T153756_pid1740.txt](https://github.com/micro-manager/micro-manager/files/5380808/CoreLog20201014T153756_pid1740.txt)

I'll admit this is likely my misunderstanding of the functionality of the adapter API. I could not find any obvious documentation describing what I am doing incorrectly.

wwarriner, 2020-10-15:
I've resolved the issue on my end, but I believe there may still be an issue due to unexpected or undocumented behavior, and am therefore leaving the issue open. I'm open to discussion about this.

To explain my resolution, my code has a C++ class representing an image, managed with a std::unique_ptr. If the unique_ptr is null when snap() is called by the Snap button or by Live, then Micro-Manager hangs. If the unique_ptr is set to an image object which yields a nullptr when GetImageBuffer() is called, Micro-Manager doesn't hang.

Coincidentally, if GetImageBufferSize() returns zero, Micro-Manager complains about insufficient memory for the circular buffer and then eventually crashes after sufficient images are acquired.

If any of this behavior is documented, I must have missed it. Was it the intent for the adapter API to expect a perpetual block of dedicated image data memory from application start to close? Perhaps I'm missing something about the design intent, and I'm happy to learn something new!

marktsuchida, 2020-10-16:
To get the easier part out of the way, `GetImageBufferSize()` should always return the expected _width_ * _height_ * _bytes-per-pixel_ given the current camera settings (as mentioned in `MMDevice.h`), even when there is no valid image. (Try `mmc.getImageBufferSize();` in the Script Panel using DemoCamera and before snapping any image.)

I don't know what is causing the hang because that depends (at least partly) on your code, but your description seems to make sense: `GetImageBuffer()` must return null if there is no image. (What _does_ it return when your `unique_ptr` is null? If it is anything but null, it won't work.) MMCore will convert the returned null to an exception being thrown.

I agree that these things need to be documented better.

Please reopen if this doesn't answer your question!

closed 2020-10-16 by marktsuchida

wwarriner, 2020-10-16:
Thank you for taking time to respond to this issue, and I appreciate your politeness. Everything you wrote makes perfect sense.

I have an assertion that the unique ptr is not null in GetImageBuffer(). Other assertions have caused the program to halt in an expected way (Windows dialog box indicating file and line). Based on this discussion it seems like the assertion is hit and the thread gets killed, but the thread manager may be unaware of this. Perhaps this is causing the hang? I can't say for certain because I only took a peek at the Java side of the application.

At any rate I have it working. If it is welcome and would be helpful I am happy to do a documentation enhancement and submit a pull request when I have some time.

marktsuchida, 2020-10-20:
@wwarriner Thanks for the update. Contributions to the documentation will be highly appreciated!