micro-manager/micro-manager #985: Improve interface for DA/AD devices

nicost, 2020-10-20:
From a small group email thread May 2020.  Suggestions from Jon Daniels that make a lot of sense to me:

I like the idea of breaking the interface/device into 2 separate device types, one for analog input and the other for analog output.  In the ASITiger device adapter we have both input and output devices.
 
Instead of simply converting the existing SignalIO interface to output-only, maybe it would be better to create two new device types and keep the existing one as-is for backward compatibility?  Then device adapters which are actively maintained could be converted to use the new device types and the old device type marked as deprecated.
 
I also concur with Nico's suggestion of renaming things but I suggest slightly different names.  How about AOSignal for analog output and then AISignal for analog input?  That matches with the NI convention of AO and AI, and also specifies that it's an analog signal.  I think it would be best to avoid "DA" in the method names; people with electronics background immediately recognize "DA" as "digital to analog" meaning it's an analog output, but that is far from obvious to everyone. 
 
So I am suggesting API methods could be

void setAOSignal(const char* signalILabel, double volt) throw (CMMError);
double getAOSignal(const char* signalILabel); // returns the voltage that was actually set on the device, which could differ from the previously set voltage
double getAOLowerLimit(const char* signalILabel) throw (CMMError);
double getAOUpperLimit(const char* signalILabel) throw (CMMError);

The sequence functions could look like:

long getAOSequenceMaxLength(const char* signalILabel) throw (CMMError);
void startAOSequence(const char* signalILabel) throw (CMMError);
void stopAOSequence(const char* signalILabel) throw (CMMError);
void loadAOSequence(const char* signalILabel,  std::vector<double> voltSequence) throw (CMMError);
 
 
Then with the analog input there would be at minimum the API method (and maybe others aren't needed)
 
double getAISignal(const char* signalILabel); // returns the voltage read from the hardware

marktsuchida, 2020-10-23:
I agree that `SignalIO` needs to be deprecated and replaced with better solutions. It is named as if it does input and output, but actually only ever did output (except where people misinterpreted `GetSignal`, which presumably was supposed to return the currently set output voltage, not any sort of input voltage -- or at least that's how I look at it, but one could also look at it the other way around). The word "signal" is also a poor choice.

Up till now, the "correct" (but poorly advertised) way to implement an analog input was as a read-only property, but it has too many limitations, because all sorts of code (like the Device Property Browser) read properties whenever they wish, whereas often we need deliberate "measurements" for analog inputs at well defined timings. Also we will likely want to add triggered and sequenced input, at least in the future, which won't work with properties. (These could be considered limitations of our property interface, but that is harder to change.)

So adding an `AnalogInput` type seems like the right thing to do. I am not 100% sure about the use of `signalLabel` (with all the management that would be needed around it). Is there a reason why we can't use a device per input channel? That way, we get label management for free (because the device label would be used), and I feel that adding another layer of labels just adds unnecessary complexity.

I would prefer the method to be named `read` or `measure` rather than `get`, because it implies an actual hardware action (AD conversion), rather than merely retrieving a (potentially cached) value.

A lot of times, people are interested in the voltage (in volts) seen by an AD converter. But applications inevitably come up where the raw AD units (8-bit, 16-bit, or whatever) are needed, such as anything that involves histogramming the input values. In addition, we need to support devices that don't know their own voltage calibration, and some AI devices support switching to current input. So I think the `readAnalogInput` method should return an unsigned (32-bit) integer, with a separate method to retrieve offset and scale factor to convert that to a voltage (or current, although that's lower priority to support). The conversion to voltage can be implemented in MMCore so that user code can read a voltage directly (using user-provided offset and scaling when the device doesn't). We also need a third method to query the number of bits.

(We may encounter AI devices that have built-in voltage calibration that is nonlinear. Supporting those will need a device method to directly read the voltage. But I don't think these are common (good ADCs and DACs are pretty darn linear compared to the rest of the world), and in any case it should be treated as the exceptional case, not the default. Otherwise we'll start seeing a lot of duplicated code in device adapters.)

Regarding analog output, I don't think we even need a dedicated device type unless we are adding new functionality: plain properties can perform all the methods listed above, including sequencing.

However, it probably makes sense to have an `AnalogOutput` device for symmetry and for better support by the `Utilities` virtual devices, among other things. Here, too, I don't see the need for a second layer of labels (the hub-peripheral mechanism already accomplishes this; see how `NIMultiAnalog` does this with `SignalIO`), and we need the primary method to use DAC units, not voltage.

jondaniels, 2020-11-20:
@marktsuchida makes some very good points.  Austin Blanco may have some good ideas based on his work with the Triggerscope but I don't think he's on GitHub.  There is a thread at https://forum.image.sc/t/new-kind-of-device-in-micro-manager/45471/7 regarding adding analog input device that accepts a stream of data.

Based on others' thoughts, here's an updated API suggestion for analog output devices:

void setAOSteps(long) throw (CMMError);  // would normally be implemented the device adapter, like stage's setPositionSteps()
long readAOSteps(); // returns the digital value that was actually set on the device
void setVoltsPerStep(double voltsPerStep);
void setVoltageOffset(double voltageOffset);
double getVoltsPerStep();
double getVoltageOffset();

void setAOVoltage(double volt) throw (CMMError);  // Device.h would provide base implementation using setAOSteps()
double readAOVoltage(); // Device.h would provide base implementation using readAOSteps

<maybe methods to get and set limits or get the bitdepth of the hardware>

long getAOSequenceMaxLength() throw (CMMError);
void startAOSequence() throw (CMMError);
void stopAOSequence() throw (CMMError);
void loadAOSequence(std::vector<double> voltSequence) throw (CMMError);

My thoughts for the analog input device aren't as well formed.  We need to accommodate different widths (those with experience with cameras will have better input than I do here).  It's worth accommodating streams of data as well as single values, whether in the same device type or a different one.  Here is an initial stab at an API.

long readAIStepsLong(); // reads a single long integer value from the hardware
std::vector<long> readAIStepLongVector(); // reads an array of long integer values (should we specify a length?)
unsigned short readAISteps16bit); // reads a single unsigned short (16 bit) integer value from the hardware
std::vector<unsigned short> readAIStes16bit(); // reads an array of unsigned short (16 bit) integer values (should we specify a length?)
unsigned char readAISteps8bit); // reads a single byte value from the hardware
std::vector<unsigned char> readAIStes8bit(); // reads an array of byte values (should we specify a length?)

AdvancedResearchConsulting, 2020-11-20:
Hi all, Jon thanks for mentioning this to me! I agree we need/want an **In device type. I'm actually getting more and more requests for this. Not to pile on but we may want to consider some form of reading in a buffer. One limitation on AI is that the measuring device may collect some frequency of samples at a higher rate than the desired ability to pull into MM. So for instance if that happened, and if the collecting system had local storage, how could we pull in the data and store it in the image metadata? Or maybe that's too big of an ask? Anyway - overall I like the idea of a dedicated outbound and inbound specification!

artmeln, 2020-11-24:
Following up on some posts on the MM forum, I think it would be nice to have an input MM device that can accommodate a general photon counting card.

A typical photon counting card streams arrays of data through a high speed port such as USB2. One record in the data array corresponds to one photon. The record contains information about the arrival time of the photon and which channel the photon was registered in. Exactly how this information is recorded depends on the manufacturer of the device but invariably it is not useful without further processing. One example of such processing is calculation of cross-correlation functions between two channels. Another is calculation of decay histograms in lifetime measurements.

Based on this, I think there is a need for a MM device that has the following minimal set of functions:

Start - tell the card to start streaming data;
Stop - tell the card to stop streaming data;
GetBuffer - get the latest data buffer from the card;
GetDataLength - get the number of events in the buffer (since it is not guaranteed to be the same every time);
GetDroppedDataFlag - read the flag that will notify the user that some data was dropped before it could be read in.

Other functions (such as setting the mode of operation or possibly time resolution) can be handled through the standard user-defined property mechanism. And when it comes to processing the streamed data and calculating something useful from it for user display, I think these can be handled through a mechanism similar to ProcessorPlugin.