micro-manager/micro-manager #987: Memory issues for 10GigE Camera Acquisitions

camFoltz, 2020-10-22:
When trying to run burst acquisitions at ~500 fps with an Oryx 10GigE Camera, I was running into what I think are memory issues.  In this test, I would run an MDA with 10,000 frames, at 0 ms interval so acquire the data as fast as possible (10 sec time series at 1ms exposure).  Micromanager is seemingly able to capture about 6000 frames but for the remaining frames, it is unable to grab the images from the camera, then MM times out, and the acquisition fails.

I am able to do this fine on a camera at the same fps but with a USB3.0 connection.  It seems to not be an issue with the buffer size, as this acquisition is only ~11 GB in total and the buffer size was set to 50 GB (64 GB RAM computer).  I also changed the Java Memory under: ImageJ > Edit > Options > Memory & Threads to 40 GB.  I also monitored both this memory and it did not ever fill the sequence buffer, but got up to 23% memory in the memory monitor.  I tried this acquisiton in pycromanager, and received similar issues where the acquisition would freeze and never complete.  It is not an issue with where the data is being saved, as there is over 1TB of free space on the disk.

For smaller acquisitions (say 500 frames) the acquisition will sometimes complete successfully, however unreliably, and in this case, it will start filling the sequence buffer. It is not an issue with the Camera, nor the 10GigE connection, as running the same acquisition in its Native software seems to work just fine.  I have attached some images of the what it looks like when the Acquisition fails.

I was wondering if anyone else has tried to take fast acquisitions with a 10GigE camera through MM and has run into similar issues.  I am happy to provide more information about the system/parameters/things I noticed during the acquisition if that will help diagnosis.

[image]

[image]

marktsuchida, 2020-10-23:
This sounds like a tough problem and I only have vague suggestions.

The fact that it works with USB3, together with the error being generated by Spinnaker, suggests that it is something happening at the driver or OS level, whereas the fact that it works with the vendor software suggests otherwise. So one possibility is that there is some long-range interaction going on due to Micro-Manager using more resources than the vendor software.

It might be worth trying with both the sequence buffer and the ImageJ memory set to _smaller_ values (just enough to not run out, and in total less than your physical RAM). These settings, when huge, can affect the layout and management of memory by the operating system in significant ways. (Although very unlikely to be related, it also happens that the Java Virtual Machine tends to perform slightly better when its memory (ImageJ memory) is set to 32 GiB or less.)

Another possibility is that Micro-Manager is using the CPU (or, more likely, memory bandwidth) too much, or simply not retrieving images from the driver fast enough because of doing too much work on the same thread in between, such that the driver's buffer is overflowing. Maybe disable display of MDA images (Tools > Options) or try running the acquisition from a script using only MMCore methods. This would have to be a subtle effect, though, because Micro-Manager should be doing the same things when using USB3. Nonetheless, it might help to get an idea of the time scale. If you decrease the frame rate instead of the frame count, at what point does the acquisition start to run reliably?

camFoltz, 2020-10-23:
conversation has been moved to [https://forum.image.sc/t/fast-time-series-acquisition-for-10gige-camera/44418/2] as it is not a direct bug issue

closed 2020-10-23 by camFoltz