micro-manager/micro-manager #994: Quick Access Panels do not re-appear on application restart

nicost, 2020-11-03:
Quick Access Panels are a nice feature making it possible for system configurators to create an easy work-flow for novice users.  However, after configuring a Quick Access Panel, it does not re-appear upon restarting MM.  That creates an unnecessary hurdle to use this feature.  We could, for instance, store which Quick Access Panels are open in the user profile and restore them on application restart.  This would greatly increase usability of Quick Access Panels.

marktsuchida, 2020-11-04:
When configuring a Quick Access Panel, there is an option "Open on launch" with options "Never", "Always", and "Remember". It would probably make sense to make "Remember" the default.

nicost, 2020-11-04:
Yes, please!

nicost, 2020-11-08:
Done in PR #999

closed 2020-11-08 by nicost